deinstapel/cpupower #135: Does not work on Gnome 3.38

mokilcde, 2020-09-17:
Hi!

After upgrade to Gnome 3.37.x and now with 3.38.0 the extensions does not show me any information
But I see this in syslog
gnome-shell[3491]: JS ERROR: TypeError: meta is null#012_patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13#012_patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18#012createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30#012CPUFreqIndicator@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:87:14#012enable/</<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:96:34#012spawn_process_check_exit_code/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:63:21

* Check if another issue with the same problem exists
  I do not see one
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
    Yes
  * policykit working and rule working
    Yes
* Check if the cpufreqctl utility works correctly
  Yes
* Please include the following information
  * CPU Model
    Intel® Core™ i7-8550U CPU @ 1.80GHz × 8
  * Linux kernel version
    Linux k-c13 5.9.0-rc4-amd64 #1 SMP Debian 5.9~rc4-1~exp1 (2020-09-11) x86_64 GNU/Linux
  * policykit version
    0.117-1
  * Gnome shell version
    3.38.0

  * journalctl -xb (maybe pastebin/filtered)
Sep 17 06:49:11 k-c13 gjs[12936]: JS LOG: [cpupower-status] ShowArrow: true
Sep 17 06:49:11 k-c13 gjs[12936]: JS LOG: [cpupower-status] ShowIcon: true
Sep 17 06:49:11 k-c13 gjs[12936]: JS LOG: [cpupower-status] UseGHzInsteadOfMHz: true
Sep 17 06:49:11 k-c13 gjs[12936]: JS LOG: [cpupower-status] ShowCurrentFrequency: true
Sep 17 06:49:11 k-c13 gjs[12936]: JS LOG: [cpupower-status] Default battery profile: None
Sep 17 06:49:11 k-c13 gjs[12936]: JS LOG: [cpupower-status] Default AC profile: None
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:11 k-c13 gnome-shell[3491]: JS ERROR: TypeError: meta is null
                                         _patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
                                         _patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
                                         createMenu@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:201:30
                                         CPUFreqBaseIndicator/<@/srv/home/michael/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:73:78
Sep 17 06:49:16 k-c13 gjs[12936]: JS LOG: [cpupower-status] Default battery profile: None
Sep 17 06:49:16 k-c13 gjs[12936]: JS LOG: [cpupower-status] Default AC profile: None

kaimast, 2020-09-24:
I just installed the extension on 3.38. 

The icon showed up at first, but when I clicked install it disappeared and I have the following stacktrace in my syslog.

```
Sep 24 16:28:34 kai-xps gnome-shell[2222]: clutter_container_remove_actor: assertion 'CLUTTER_IS_CONTAINER (container)' failed
Sep 24 16:28:34 kai-xps gnome-shell[2222]: == Stack trace for context 0x561ae20b5220 ==
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #0   561ae7afb188 i   /home/kai/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:100 (a39248b7560 @ 31)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #1   561ae7afb100 i   /home/kai/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:108 (def02765bf0 @ 29)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #2   561ae7afb028 i   resource:///org/gnome/shell/ui/extensionSystem.js:109 (49b5a938c90 @ 395)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #3   561ae7afaf98 i   resource:///org/gnome/shell/ui/extensionSystem.js:502 (49b5a9384c0 @ 15)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #4   7ffe2c754ec0 b   self-hosted:225 (6d0c0ba5e70 @ 273)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #5   561ae7afaf08 i   resource:///org/gnome/shell/ui/extensionSystem.js:502 (49b5a938600 @ 120)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #6   7ffe2c755ae0 b   self-hosted:850 (a39248130b0 @ 454)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #7   7ffe2c7573d0 b   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:533 (6d0c0b90100 @ 169)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #8   561ae7afae58 i   resource:///org/gnome/shell/ui/extensionSystem.js:210 (49b5a938b50 @ 200)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #9   561ae7afadc8 i   resource:///org/gnome/shell/ui/shellDBus.js:311 (2bc4f8493ce0 @ 20)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #10   561ae7aface0 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:327 (6d0c0b90830 @ 189)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #11   561ae7afac28 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:404 (6d0c0b906f0 @ 34)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: == Stack trace for context 0x561ae20b5220 ==
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #0   561ae7afb188 i   /home/kai/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/baseindicator.js:107 (a39248b74c0 @ 22)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #1   561ae7afb100 i   /home/kai/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:109 (def02765bf0 @ 45)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #2   561ae7afb028 i   resource:///org/gnome/shell/ui/extensionSystem.js:109 (49b5a938c90 @ 395)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #3   561ae7afaf98 i   resource:///org/gnome/shell/ui/extensionSystem.js:502 (49b5a9384c0 @ 15)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #4   7ffe2c754ec0 b   self-hosted:225 (6d0c0ba5e70 @ 273)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #5   561ae7afaf08 i   resource:///org/gnome/shell/ui/extensionSystem.js:502 (49b5a938600 @ 120)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #6   7ffe2c755ae0 b   self-hosted:850 (a39248130b0 @ 454)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #7   7ffe2c7573d0 b   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:533 (6d0c0b90100 @ 169)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: Object .Gjs_ui_panelMenu_PanelMenuButton (0x561ae2dbbc70), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #8   561ae7afae58 i   resource:///org/gnome/shell/ui/extensionSystem.js:210 (49b5a938b50 @ 200)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #9   561ae7afadc8 i   resource:///org/gnome/shell/ui/shellDBus.js:311 (2bc4f8493ce0 @ 20)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #10   561ae7aface0 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:327 (6d0c0b90830 @ 189)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: #11   561ae7afac28 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:404 (6d0c0b906f0 @ 34)
Sep 24 16:28:34 kai-xps gnome-shell[2222]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
```

fin-ger, 2020-09-30:
Sorry for the long delay. But for this issue to be addressed I have to wait for Fedora to ship Gnome 3.38. Sorry for the long delay.

@martin31821 Has Arch shipped 3.38 already? I think all other contributors moved to other distros since the last major release of this extension...

martin31821, 2020-09-30:
3.38 is still in staging here, I'm on 3.36.6 at the momen. 
When it's released, I can take a look, otherwise I only can merge PRs.

fin-ger, 2020-09-30:
So then let's wait for 3.38 to land in the major distros.

mokilcde, 2020-10-03:
This patch works for me

```
diff -pbaur /tmp/indicator.js src/indicator.js
--- /tmp/indicator.js	2020-10-03 07:28:19.590913282 +0200
+++ src/indicator.js	2020-10-03 07:29:13.111214991 +0200
@@ -191,6 +191,7 @@ var CPUFreqIndicator = class CPUFreqIndi
 
         this.imSliderMin = new PopupMenu.PopupBaseMenuItem({activate: false});
         this.minSlider = new Slider.Slider(this.minVal / 100);
+        this.minSlider.x_expand = true;
         this.minSlider.connect(parseFloat(Config.PACKAGE_VERSION.substring(0,4)) > 3.32 ? 'notify::value' : 'value-changed', item => {
             this.minVal = Math.floor(item.value * 100);
             this.imMinLabel.set_text(this._getMinText());
@@ -198,13 +199,14 @@ var CPUFreqIndicator = class CPUFreqIndi
         });
 
         if (parseFloat(Config.PACKAGE_VERSION.substring(0,4)) > 3.32) {
-            this.imSliderMin.add(this.minSlider, {expand: true});
+            this.imSliderMin.add_child(this.minSlider);
         } else {
             this.imSliderMin.actor.add(this.minSlider.actor, {expand: true});
         }
 
         this.imSliderMax = new PopupMenu.PopupBaseMenuItem({activate: false});
         this.maxSlider = new Slider.Slider(this.maxVal / 100);
+        this.maxSlider.x_expand = true;
         this.maxSlider.connect(parseFloat(Config.PACKAGE_VERSION.substring(0,4)) > 3.32 ? 'notify::value' : 'value-changed', item => {
             this.maxVal = Math.floor(item.value * 100);
             this.imMaxLabel.set_text(this._getMaxText());
@@ -212,7 +214,7 @@ var CPUFreqIndicator = class CPUFreqIndi
         });
 
         if (parseFloat(Config.PACKAGE_VERSION.substring(0,4)) > 3.32) {
-            this.imSliderMax.add(this.maxSlider, {expand: true});
+            this.imSliderMax.add_child(this.maxSlider);
         } else {
             this.imSliderMax.actor.add(this.maxSlider.actor, {expand: true});
         }
```

fin-ger, 2020-10-03:
Many thanks for patching this! This helps a lot!

However, before applying this to cpupower, we have to make sure that this does not create issues on older Gnome platforms until 3.28. As I currently am lacking time to test for this: can anyone else test through Ubuntu LTS versions, latest Fedora, latest Arch, and Debian stable?

mastercaution, 2020-10-04:
Nice patch! ... I guess. I tested it quickly on:

- [x]  latest Arch (GNOME 3.36.6)
- [x] Ubuntu 20.04
- [x] Ubuntu 19.10
- [x] Ubuntu 18.04

I didn't find any issues.

lviggiani, 2020-10-05:
Hi I've just received GNOME 3.38 on Arch and at first the extension wasn't working anymore but after applying the patch by @mokilcde it now forks fine, thanks!

Cervoxx, 2020-10-06:
Patch works here, gnome 3.38 Endeavour OS/Arch. But boy it was a process to learn how to modify this AUR PKGBUILD to apply it: https://aur.archlinux.org/packages/gnome-shell-extension-cpupower-git/ 

[gnome-shell-extension-cpupower-git.zip](https://github.com/martin31821/cpupower/files/5335105/gnome-shell-extension-cpupower-git.zip)
This is what I came up with, but its most likely not expertly done.

EmmanuelCharpentier, 2020-10-07:
I've recently been bitten by this one on Debian testing. `mokilcde` patch worked for me.

mastercaution, 2020-10-07:
@mokilcde Would you like to create a PR implementing your patch, so we can get this extension running on 3.38? Thanks :D

mokilcde, 2020-10-07:
Done: #137

martin31821, 2020-10-08:
Thanks for preparing, I'll upload a new release today, so it should be available by the end of the week!

jawse, 2020-10-16:
Has this been updated on the gnome extensions site.  Still not working for me on Fedora 33?

jawse, 2020-10-16:
NVM, i installed from the latest src and its working perfect.  Thank you.

martin31821, 2020-10-17:
The update is not yet on the gnome extensions page, sorry.
There was some delay on my side.

Cervoxx, 2020-10-18:
https://aur.archlinux.org/packages/gnome-shell-extension-cpupower-git/ This aur package has been a godsend in the meantime.

martin31821, 2020-10-30:
Update on extensions.gnome.org was published and approved.

closed 2020-10-30 by martin31821