deinstapel/cpupower #136: Settings reset after i wake up my computer from suspend

zdnearchou, 2020-10-05:
I set my power setting to Battery Saver so it works with BCLK x8 with no turboboost after i suspend my computer and wake it up, despite that the settings remain the same, the frequency of the cpu goes up to the base clock (x18) until i switch a profile then switch it back to the battery saver again.

Thanks in advance,

Additional information:

Lenovo X1 ThinkPad 2nd
i5 4300U
Linux Kernel: 5.4.0
Polkit : 0.105
Gnome 3.36.3
journalctl: https://pastebin.com/901AjG1K

mastercaution, 2020-10-11:
Do you have configured another profile in the auto switching configuration? Because if your device wakes up, cpupower may change the selected profile according to your auto switch configuration (only if this feature is turned on).

ioanmoldovan95, 2021-06-13:
I have a similar problem, with a Lenovo Thinkpad T450s, i7-5600U.
I have 2 profiles set, one with full cpu usage (max freq 100, turbo boost on), called AC, and one with 50% max cpu freq and turbo boost off as Battery, auto switch on.
Everything works as expected until laptop wakes from sleep, after that it seems to disregard any settings I do in cpu power manager, frequencies go up to 3.2GHz (which is max turbo boost frequency), even though "cat /sys/devices/system/cpu/intel_pstate/no_turbo" outputs 1...

OS: Pop os 21.04
Linux Kernel: 5.11.0
CPU: i7-5600U
Gnome: 3.38.5

fin-ger, 2021-06-13:
This sounds like a kernel issue then. As far as I understand it, you are able to reproduce the issue without this extension solely based on the intel-pstate kernel API? If yes, would you consider reporting this issue directly to the Linux kernel developers?

fin-ger, 2021-09-27:
Any news on this?

ioanmoldovan95, 2021-11-11:
Sorry, I sold the laptop a bit of time after leaving that comment, so I no longer have the means of testing anything. Thanks for the response