deinstapel/cpupower #138: Because of high battery drain: Option to make the extension not poll CPU frequency

ruffson, 2020-10-15:
I noticed that this extension uses up quite a lot of energy when I am on battery. With the extension enabled I have a discharge rate of about 15-23W and without it I go down to about 8W. (I think this is due to `intel_atomic_commit_work`?)

And because I use this extension mainly to disable the boost and to switch profiles I would like to disable polling for the CPU frequency. Afaik switching profiles and disabling boost is just a matter of changing some files once per change, which does not require constant polling. I already disabled showing current frequency but it still shows in the menu when I click on it.

Really like the extension otherwise!

mastercaution, 2020-10-15:
Thanks for your suggestion!
You're right, the extension polls the frequency regardless of your setting. The setting only decides whether it should be displayed or not. But it only polls once per second ... I don't think that this can cause a 100% increase in power usage.

However, we will definitely work on this, because the performance junkie inside me cannot accept that waste of cycles :joy:

fin-ger, 2020-10-16:
Hi @ruffson,

thank you for bringing this up. I just checked on my system whether disabling cpupower changes the overall battery consumption of my system. I cannot confirm this, but I also have a whole lot of other extensions running (e.g. Freon) which also do periodical updating.

The highest battery consumption on my system is created by `tick_sched_timer` with ~1W. I checked this with `powertop`.

`intel_atomic_commit_work` is at ~40mW on my system.

For reference: [this is the glib function we are using for periodic updating](https://developer.gnome.org/pygobject/stable/glib-functions.html#function-glib--timeout-add-seconds). It already aims to be battery efficient.

Just to check whether it is really related to the periodic updating: Can you uncomment [lines 101 and 102 in `indicator.js`](https://github.com/martin31821/cpupower/blob/master/src/indicator.js#L101) in your local cpupower installation and check with `powertop` if the energy consumption changes? The file is most likely located at `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js`.

@mastercaution disabling the periodic checking will most likely not change anything, as we must check for CPU configuration changes made outside of this extension to prevent bugs like #9. As we are using the `timeout_add_seconds` function, the periodic callbacks for frequency checking and configuration checking are done simultaneously. Polling for the frequency should not create extra noise.

ruffson, 2020-10-16:
Thank you @fin-ger and @mastercaution for your help!

I commented out those lines. It did not seem to change things. With or without these lines commented out I get about 9W currently with everything else being equal and after letting it settle down...

So your extension was not at fault. I am sorry for blaming you guys on my power consumption problems :)
I just bought a new computer and many users reported quite low idle battery drain and I observed a much higher drain rate. I disabled extension by extension and monitored the changes in `powertop`. And it seemed that this extension was one big part of my high power consumption. 

However it is quite hard to test these things because powertop's low refresh rate (which makes sense I guess to reduce load from its own measurements) and only rough estimates on power consuming processes. And since `tick_sched_timer` and `intel_atomic_commit_work` came up frequently among the Top 5 consuming processes with this extension and lower total power consumption when I disabled it, it seemed clear and plausible that it was caused by this.

Sorry guys. However for people who just want to set the profiles and boost I still think it would be best to limit continous operations from this extension with an optional setting. But @fin-ger you are saying that you need to make checks to show the current state? Is the current state data taken by some system process and cached so you can just read out instead of asserting calls to the CPU directly?

fin-ger, 2020-10-16:
> But @fin-ger you are saying that you need to make checks to show the current state?

Yes, it's done [here](https://github.com/martin31821/cpupower/blob/13dfdf2ef1475b757f9607faf432c202c3e74e49/src/indicator.js#L379).

> Is the current state data taken by some system process and cached so you can just read out instead of asserting calls to the CPU directly?

We are using the [sysfs controls by the intel pstate driver](https://www.kernel.org/doc/Documentation/cpu-freq/intel-pstate.txt) (e.g. [turbo boost state](https://github.com/martin31821/cpupower/blob/master/tool/cpufreqctl#L144)). It essentially makes the CPU frequency and turbo boost state readable from the filesystem. We are using the [`cpufreqctl` helper script](https://github.com/martin31821/cpupower/blob/master/tool/cpufreqctl) to read these values and pass them on to the cpupower extension.

**TL;DR:**
We are not implementing any kind of intel pstate driver on our own. We are using the Linux kernel sysfs controls. The intel pstate driver is included in any modern Linux kernel.

ruffson, 2020-10-16:
> **TL;DR:**
> We are not implementing any kind of intel pstate driver on our own. We are using the Linux kernel sysfs controls. The intel pstate driver is included in any modern Linux kernel.

Thank you for clarifying!

closed 2020-10-16 by ruffson

mastercaution, 2020-10-16:
> @mastercaution disabling the periodic checking will most likely not change anything, as we must check for CPU configuration changes made outside of this extension to prevent bugs like #9. As we are [...]

My thought was that, if the popup menu is closed and no frequency is displayed, we don't need to poll anything

fin-ger, 2020-10-16:
@mastercaution Ah, yes that's absolutely the best option. Totally missed you on that point!