deinstapel/cpupower #139: Missing icon on taskbar in Gnome version 3.38

aniketaayush, 2020-10-24:
I installed the extension from gnome extensions but the icon in the taskbar is not there. I verified, everything is installed properly in my system and the app works fine when launched through extensions. Others have also faced a similar issue with Gnome version 3.38 ( See comments at [https://extensions.gnome.org/extension/945/cpu-power-manager/](extension page). It's a very handy tool and I have been using it for 1 year. Kindly fix the issue. Thanks.

lucaghislo, 2020-10-25:
I got the same issue since updating to Gnome 3.38 on Pop!_OS 20.10. I've been using this extension for months now and it saved me so many times. Hope the issue will be fixed soon.

fin-ger, 2020-10-25:
I think none of the maintainers here is on 3.38, yet. We will wait for 3.38 to land in our distros. Instead of waiting for us, you can also try to fix this issue on your own. We are happy to merge any pull requests!

aniketaayush, 2020-10-25:
Will building it manually fix the issue since someone told the fix is already there in the master branch ?

lucaghislo, 2020-10-25:
I've just seen indicator.js has been modified to fix the whole 3.38 issue, is that correct?

aniketaayush, 2020-10-25:
Also yesterday, I built it manually on my pc following all the instructions mentioned but the extension didn't show up in my extension list. Anything extra I need to do? (I executed the scripts also)

fin-ger, 2020-10-25:
Yes, we merged some fixes for 3.38. I just assumed they were already live on gnome.extensions.org but this does not seem to be the case, sorry. We issued a new version on the extension website 17 days ago and are waiting for the Gnome Extensions maintainers to approve the new version. It will go live when the Gnome Extensions maintainers approve the new version.

In the meantime, a manual build will fix the issue for you. Don't hesitate to contact us, if you have any questions regarding the manual install!

lucaghislo, 2020-10-25:
Excellent, really appreciate the help! Thanks ;)

aniketaayush, 2020-10-25:
@fin-ger Plz brief me on the process because I tried everything mentioned in the instructions but the extension is not showing up in my extensions. Thanks.

fin-ger, 2020-10-25:
> Also yesterday, I built it manually on my pc following all the instructions mentioned but the extension didn't show up in my extension list. Anything extra I need to do? (I executed the scripts also)

If you are running a Gnome on a Wayland session, you have to re-login after installing any extension manually :disappointed: 

You can check the gnome-shell logs after login like this:

```
journalctl /usr/bin/gnome-shell -f
```

It will not filter for only cpupower, but show logs for all extensions on your system. Read through it and look if cpupower is somewhere in it.

aniketaayush, 2020-10-25:
I did re-login after installation still it wasn't showing up.

fin-ger, 2020-10-25:
To mimic the installation process of the extension website, use the instructions for [Installing for another (admin) user](https://github.com/martin31821/cpupower#installing-for-another-admin-user).

Which manual installation method have you used before?

aniketaayush, 2020-10-25:
I have followed the same.

fin-ger, 2020-10-25:
Hmm, it looks like we missed to add the new gnome-shell version to the metadata.json file. Can you check `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/metadata.json` and see if 3.38 is listed? If not, just insert it manually and check if the extensions loads now.

aniketaayush, 2020-10-25:
After installing the extension and running the install script, I need to re-login and edit the JSON file, right?

fin-ger, 2020-10-25:
edit the json file first and then re-login

aniketaayush, 2020-10-25:
Ok, so I don't have any folder with the name cpupower in the extensions folder. What's the issue?

aniketaayush, 2020-10-25:
@fin-ger Thanks a lot for your help it's working properly now :) . For others who are facing the same issue,use this workaround :- 
1.>Install the extension manually by following the instructions mentioned in the readme file (make sure to replace username with your actual username).
2.>Go to `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/tool` and run installer.sh using `sudo ./installer.sh install` .
3.> Edit the `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/metadata.json` file and add 3.38 to the shell-version .
4.> Restart the shell by pressing 'Alt+F2', entering r in the command field, and hitting enter. (No need to re-login)
5.>Enable the extension from settings.
6.>Done :smile:

fin-ger, 2020-10-25:
Some hints on this:

`sudo ./installer.sh install` should not be necessary and can be a security issue when the environment variables are not properly configured.

`Alt+F2 r` is only working on X11, it is not working on Wayland sessions.

@mastercaution @unfug can you update the metadata.json and upload a new version to the extensions website?

aniketaayush, 2020-10-25:
Oh, I did run the installer.sh using sudo, what all issues can it create ?

fin-ger, 2020-10-25:
It behaves differently when the TOOL_SUFFIX is set:

https://github.com/martin31821/cpupower/blob/master/tool/installer.sh#L98

It decides whether to install as a system application or as a user installation. When installed as a system tool, any user on your system can get root access if the location of the cpufreqctl binary is writable by them.

Check /usr/share/polkit-1/actions and /usr/share/polkit-1/rules for cpupower files.

Also check `/usr/bin` and `/usr/local/bin` for cpufreqctl scripts. All scripts, rules, and actions should have your username in the name or file content. Otherwise it's a system wide installation, which you are not targeting. Just delete the cpupower files without your username in it.

fin-ger, 2020-10-25:
Gaining root access is also possible for any app you run as your user.

aniketaayush, 2020-10-25:
Can some script be run to remove such behavior? Or is there any simpler way than going in each file and deleting?

aniketaayush, 2020-10-25:
Also, I just ran the command `sudo ./installer.sh install', nothing else

fin-ger, 2020-10-25:
```
sudo ./installer.sh uninstall
```

should do the trick. But to be sure you can check manually. Should only be 3 files.

aniketaayush, 2020-10-25:
I have 3 files named konkor.cpufreq.policy, and 2 entries for  mko.cpupower.setcpufrequency in policy folder.One entry has my username in the filename whereas one doesn't. Should I delete that?

aniketaayush, 2020-10-25:
@fin-ger Ok so I ran the uninstall command and after that, the mko.cpupower.setcpufrequency policy which didn't had my username got deleted. I also checked `/usr/bin` and `/usr/local/bin` and '/usr/local/bin` contains a single script named cpufreqctl-username. So now, there is no security issue, right ?

fin-ger, 2020-10-25:
Correct. You can additionally check with `ls -l` if your user has write access to `/usr/local/bin/cpufreqctl-username`. If not, everything is fine.

aniketaayush, 2020-10-25:
@fin-ger `-rwxr-xr-x 1 root root 6038 Aug 23 23:26 cpufreqctl-username` is the output I got for the script.Is it ok ?

fin-ger, 2020-10-25:
Yes, only root can write. Others cannot even read it, just execute it. It's like it should be :+1:

aniketaayush, 2020-10-25:
@fin-ger Thanks a lot for your help. Also, I have edited my workaround comment to include your inputs. Just one last thing, what u mean by just execute it in your last comment(like should I execute it now or you wanted to tell that others can execute it) ?

fin-ger, 2020-10-25:
The script can be executed by your user by the gnome extension. No need to run it manually.

aniketaayush, 2020-10-25:
Got it, thanks. Also, should I close the issue?

fin-ger, 2020-10-25:
We'll close it when the metadata issue is fixed on the extension website.

maxjf1, 2020-10-26:
Getting this issue too, when updated to ubuntu 20.10

System info:
```bash
                         ./+o+-       maxjf1@maxjf1-G3-3579
                  yyyyy- -yyyyyy+      OS: Ubuntu 20.10 groovy
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 5.8.0-25-generic
           .++ .:/++++++/-.+sss/`      Uptime: 2h 17m
         .:++o:  /++++++++/:--:/-      Packages: 2172
        o:+o+:++.`..```.-/oo+++++/     Shell: zsh 5.8
       .:+o:+o/.          `+sssoo+/    Resolution: 1920x1080
  .++/+:+oo+o:`             /sssooo.   DE: GNOME 3.38.1
 /+++//+:`oo+o               /::--:.   WM: Mutter
 \+/+o+++`o++o               ++////.   WM Theme: Adwaita
  .++.o+++oo+:`             /dddhhh.   GTK Theme: Yaru-dark [GTK2/3]
       .+.o+oo:.          `oddhhhh+    Icon Theme: WhiteSur-dark
        \+.++o+o``-````.:ohdhhhhh+     Font: Garuda 11
         `:o+++ `ohhhhhhhhyo++os:      Disk: 53G / 69G (81%)
           .o:`.syhhhhhhh/.oo++o`      CPU: Intel Core i5-8300H @ 8x 4GHz [44.0°C]
               /osyyyyyyo++ooo+++/     GPU: GeForce GTX 1050
                   ````` +oo+++o\:     RAM: 6284MiB / 15854MiB
                          `oo++.      

```

GJRTimmer, 2020-10-26:
Same issue

```bash
             /////////////                gtimmer@aeris 
         /////////////////////            ------------- 
      ///////*767////////////////         OS: Pop!_OS 20.10 x86_64 
    //////7676767676*//////////////       Host: N85_N87,HJ,HJ1,HK1 
   /////76767//7676767//////////////      Kernel: 5.8.0-7625-generic 
  /////767676///*76767///////////////     Uptime: 17 mins 
 ///////767676///76767.///7676*///////    Packages: 2415 (dpkg), 50 (flatpak) 
/////////767676//76767///767676////////   Shell: bash 5.0.17 
//////////76767676767////76767/////////   Resolution: 1920x1080, 1920x1080, 2560x1440 
///////////76767676//////7676//////////   DE: GNOME 3.38.1 
////////////,7676,///////767///////////   WM: Mutter 
/////////////*7676///////76////////////   WM Theme: Pop 
///////////////7676////////////////////   Theme: Pop-dark [GTK2/3] 
 ///////////////7676///767////////////    Icons: Pop [GTK2/3] 
  //////////////////////'////////////     Terminal: gjs 
   //////.7676767676767676767,//////      CPU: Intel i7-7700HQ (8) @ 3.800GHz 
    /////767676767676767676767/////       GPU: NVIDIA GeForce GTX 1050 Ti Mobile 
      ///////////////////////////         GPU: Intel HD Graphics 630 
         /////////////////////            Memory: 2765MiB / 31998MiB 
             /////////////
```

fin-ger, 2020-10-26:
wow... I hate this auto-close feature :angry:

fin-ger, 2020-10-26:
Can someone with Gnome 3.38 please test with the ["Installing for another (admin) user"](https://github.com/martin31821/cpupower#installing-for-another-admin-user) installation instructions?

llopisga, 2020-10-26:
@fin-ger I have just tested it for Gnome 3.38.1 and it works. For me, there was no need to log out,  it appeared just with "Alt+F2 r". Thanks!

fin-ger, 2020-10-26:
I just created a new release.

@martin31821 Can you upload the new version to the extension website?

martin31821, 2020-10-27:
@fin-ger done

GJRTimmer, 2020-10-27:
@martin31821 How long is normally the waiting period for the new version to show up on the extension website ?

martin31821, 2020-10-27:
@GJRTimmer one day to one week, usually. There is a manual approval/reviewprocess from one of the maintainers, which we now depend upon.

cdysthe, 2020-10-27:
Having the same issue on 20.10. Manual install solved the problem.

foldfelis, 2020-10-28:
Having the same issue on Endeavoros. Solved by installing the master version.

martin31821, 2020-10-30:
Upstream gnome shell should work again, was approved today.

commit 5c7020e1 (2020-10-26): Add Gnome 3.38 to metadata - this might fix #139 and #135
closed 2020-10-30 by martin31821