deinstapel/cpupower #144: CPU Locked at 800MHz

larovere, 2021-01-09:
CPU is stuck permanently at 800MHz. The issue persists regardless of setting and even after uninstalling via Gnome extensions

  * CPU Model: Core i7 8th gen
  * Linux kernel version: 5.8.0-25-generic
  * Gnome shell version: 3.38.1

mastercaution, 2021-01-09:
Sorry that you are experiencing this issue. The pstate configuration made by the extension should be discarded after a restart. You can check this by manually reading the pstate configuration with the following commands:

* `cat /sys/devices/system/cpu/intel_pstate/min_perf_pct` to get the minimum performance setting. It should be something around 0 (8 in my case)
* `cat /sys/devices/system/cpu/intel_pstate/max_perf_pct` to get the maximum performance setting. Usually it is 100.

larovere, 2021-01-10:
I did run those, and I get 19 and 100, so that seems right. Even so, monitoring the CPU with  `watch -n.1 "cat /proc/cpuinfo | grep \"^[c]pu MHz\""` shows it still remains locked at 800MHz. I should have mentioned this in the original post, but this does continue after rebooting. 

I tried to include what may be useful here:
To make sure there isn't an issue with the hardware, I booted into the other linux os I have installed and the CPU does seem to work normally. 

If I use the uninstall option in the settings of the add-on, it crashes.

From issue #48, it looks like the only thing I would need to remove is /usr/share/polkit-1/actions/mko.cpupower.setcpufreq.policy, but no such file exists

Output of `journalctl -xb` https://pastebin.com/XYZvWDD0

Policykit version: I could not figure out how to determine this

Do you have any further ideas on what to do?

mastercaution, 2021-01-11:
Thanks for the details! 
Hmm, that doesn't sound good. The polkit policy file in the newer versions can be at various places with different names depending on the installation method used. If installed via Gnome extensions by pressing the "Attempt installation" button the policy file is created at `/usr/share/polkit-1/actions/mko.cpupower.setcpufreq.USERNAME.policy`. This file should be deleted by pressing the "Uninstall" button in the extensions preferences. However, I don't think that keeping this file after removing the extension can cause this issue.

larovere, 2021-01-12:
Yes, you were right about '/usr/share/polkit-1/actions/mko.cpupower.setcpufreq.USERNAME.policy' 

It was a fresh install, so I assumed wrongly that the issue was with the extension, as it was the only thing I had added. Since the issue persists after a reinstall, it seems to be unrelated.

I appreciate the help you offered. I apologize, as I was apparently way off about what is causing this.

mastercaution, 2021-01-12:
No problem! We also appreciate your support in improving the extension.

ioanmoldovan95, 2021-06-13:
@larovere check what cpu governor your distribution uses, I found out that my processor locks at 800mhz if i use powersave governor...

fin-ger, 2021-09-27:
@larovere Does this issue still persist?

larovere, 2021-09-29:
@ioanmoldovan95 Thank you for the suggestion. I had tried that but it didn't help in my case

@fin-ger I tried every everything that seemed remotely plausible to fix the CPU issues, but the laptop died not long after. My best guess now is that it was related to the hardware failing

fin-ger, 2021-09-29:
Ah, I'm sorry your laptop died, I hope you could rescue all your important data and were able to get a replacement. I'll close this issue as the failure seems not to be related to cpupower.

closed 2021-09-29 by fin-ger