deinstapel/cpupower #145: Add support for GNOME 40

frandieguez, 2021-02-03:
**Extension Information**

- Shell Version: GNOME 40
- Extension Version: 23
- Install method: extension.gnome.org
- OS: Fedora 33 with GNOME 40 alfa installed

```sudo dnf copr enable haeckerfelix/gnome-shell-40 && sudo dnf update```

**Describe the bug**

GNOME 40 is already in alpha and it forces the usage of GTK4 for extensions dialogs, please consider adding support for it.

The extensions seems to work so far but when you try to open the preferences dialog of your awesome extension it crashes with the message below 👇

```
Gtk.BuilderError: /home/fran/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:88:1 Invalid object type 'GtkButtonBox'

Stack trace:
  CPUPowerPreferences@/home/fran/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:53:22
  buildPrefsWidget@/home/fran/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:222:19
  start@resource:///org/gnome/gjs/modules/script/package.js:206:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
  
```

Please update you extension to keep enjoying using it. Thanks for your incredible work!

martin31821, 2021-02-03:
Gnome 4 will most likely be supported after it hits the large distributions. At the moment I'm not able to update yet.

fin-ger, 2021-02-04:
I wont have time to work on this until April. Also, is it really necessary to temper with my own system to develop for a new Gnome platform? I remember that I was able to run another version of the Gnome Shell within a window in my production setup some time ago. I just can't remember how I managed to achieve this...

frandieguez, 2021-02-06:
You can test it from GNOME Boxes by selecting the GNOME OS Nightly image

fin-ger, 2021-02-06:
Hmm, intel p-state will not be available, but I guess this can be mocked with some hacks. As I'm on AMD Ryzen now, I might integrate support for zenstates or ryzenadj along with Gnome 40 support in April (I still have my old intel machine for testing). I might also look into the `/sys/devices/system/cpu/cpufreq/` kernel-API, maybe it's a more general approach which might serve this extension well. I've not decided, yet.

mlaverdiere, 2021-04-03:
For reference, in case it might be useful, here's the guide to port extensions to GNOME 40: https://gjs.guide/extensions/upgrading/gnome-shell-40.html#metadata-json

Thanks for this extension.

mastercaution, 2021-04-08:
Gnome 40 landed on Arch yesterday. This issue still exists. 
We need to port our GTK3 glade file to GTK4. The `gtk4-builder-tool` (Reference: https://developer.gnome.org/gtk4/stable/gtk4-builder-tool.html) might come in handy as well as the guide suggested by @mlaverdiere.

The output of `gtk4-builder-tool simplify --3to4 data/cpupower-preferences.glade` looks like a bit of work, and unfortunately I'm pretty busy at the moment.

To get everything else than the preferences window working we just have to add the new Gnome version to the `metadata.json` like described [here](https://gjs.guide/extensions/upgrading/gnome-shell-40.html#metadata-json).

fin-ger, 2021-04-10:
I'll look into this as soon as I can, currently still busy with my master defense.

I personally think that Gtk4 was pulled into Gnome Extensions too early, as libadwaita is not available yet. I'm less motivated to rebuild the extensions to Gtk4 just to rebuild it again for Gnome 42... So probably the Gnome40 release of this extension will provide less stability and comfort than other releases of cpupower. I'm not quite sure how to handle this properly, I'm also not sure if Gtk4 would run on Ubuntu 18.04/etc. I would love to still support those old Gnome releases with the latest version of this extension as people are still using it. At least Gnome 3.36 should be supported by the latest release. However, I don't know if we can target Gtk4 on these platforms... Maybe there is a way to let Gnome40 run Gtk3 until Gtk3 slowly disappears from distros and Gtk4 can be assumed to be available on older platforms. Does anyone know more about this?

pszafer, 2021-04-10:
@fin-ger if you target only new release in manifest.json then it should install older version of extension in older Gnome and the newest version in Gnome 40

fin-ger, 2021-04-10:
Okay, but this would remove support for all other Gnome versions in the latest release. I'm myself still on 3.36 as 3.38 is too unstable to get my daily work done and 40 is not available yet. I would prefer not to ditch all previous Gnome versions.

keeferrourke, 2021-04-28:
GNOME 40 is officially available through the Fedora 34 release that was made widely available today.

If you need volunteers to get this extension ported, I am dearly missing it in my newly upgraded shell ;)

bcclsn, 2021-05-04:
so, then, is there any news?

fin-ger, 2021-05-04:
Working on it, but may take some time to finish. Maybe next week or sth.

fin-ger, 2021-05-10:
Some news on this: the Gnome OS installation has a broken polkit setup where pkexec isn't able to prompt for user authentication. Therefore, the complete extension is unable to operate. I am now looking for an alternative to develop and try out the Gnome 40 support. Apart from that, Gnome OS seems like a bad choice as I wasn't able to get a dev environment set up (git, shellcheck, make, etc.), leading to continuous copying of new code from the host system paired with the everlasting log-out/log-in procedure to reload the extension code... 

The dev experience of Gnome gets worse and worse with every new release....

shroudedcode, 2021-05-10:
> ...paired with the everlasting log-out/log-in procedure to reload the extension code...

Is there a reason you can't use <kbd>Ctrl</kbd>+<kbd>F2</kbd> + `r` + <kbd>Enter</kbd>? 🤔

fin-ger, 2021-05-10:
> > > Is there a reason you can't use Ctrl+F2 + `r` + Enter?
> > 
> > 
> > Because it is not available on Wayland
> 
> Ah, that's a bummer, but maybe not even an issue in this particular case since @fin-ger is already using a virtual machine for development and could [disable Wayland](https://wiki.archlinux.org/title/GDM#Use_Xorg_backend) there. This shouldn't cause any issues as the extension doesn't contain any Wayland/X11-specific code (correct me if I'm wrong).

I already tried that, in Gnome OS selecting `GNOME on Xorg` crashes the session and throws you back to the login screen (running in Gnome Boxes). But, I'm currently running Fedora 34 in a VM, let's see how this goes...

fin-ger, 2021-05-11:
Well... I invested 2 days in getting a GTK4 preferences window running (I barely managed to get it open). This is mostly due to missing resources on the web (as GTK4 is quite new) and missing glade support which forces me to write ui files by hand. I got tired of it and left it aside, also because most changes were breaking for my current 3.36 setup.

Just right now I hacked together a tiny wrapper for the `prefs.js` file which is causing the whole problem of forcing me to use GTK4. I polyfilled the `imports.misc` calls and managed to get the GTK3 preferences dialog running and fully functional on Fedora 34 with Gnome 40 in ~45 lines of js. This solution feels really hacky, but it works... so might be worth some testing I guess?

I will assemble a fully working version tomorrow and commit it to git, right now it's a bunch of randomly stitched scripts in `lg` and an interactive `gjs` shell xD

fin-ger, 2021-05-17:
Successfully rebased Gnome 40 support on master. Now includes changes of #142. Will prepare a PR now

fin-ger, 2021-05-17:
Part of #151

stefangweichinger, 2021-05-20:
I installed the current version via cloning and "make install ...", in a Fedora 34 / GNOME 40 setup.

The addon works, I can use the slider from #142, but clicking on "Settings" fails with:

```
The settings of extension cpupower@mko-sl.de had an error:

Gtk.BuilderError: /home/sgw/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:85:1 Invalid object type 'GtkButtonBox'
```

Stack trace:
```
CPUPowerPreferences@/home/sgw/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:54:22
buildPrefsWidget@/home/sgw/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

I thought I should report that back to you. Thanks.

fin-ger, 2021-05-20:
@stefangweichinger It looks like you're not on the `feature/modular-cpufreqctl-backends` branch. The master branch has no support for Gnome 40 yet, the PR #151 is still pending.

fin-ger, 2021-05-20:
We're currently looking for reviews for #151 regarding the source code and compatibility with the major distros. There have been found some major issues in Gnome <3.34 installs already, but I haven't had the time to fix these issues yet. Feel free to look at the code and give feedback :D

stefangweichinger, 2021-05-20:
> @stefangweichinger It looks like you're not on the `feature/modular-cpufreqctl-backends` branch. The master branch has no support for Gnome 40 yet, the PR #151 is still pending.

Ah, right, I installed from that branch now, and settings work now. Great, thanks. Sorry for the noise ;-)

fin-ger, 2021-05-20:
@stefangweichinger Please note that when installing from e.g. feature branches you are required to uninstall the tool and polkit rules and reinstall it to make the extension work flawlessly. The update procedure for polkit rules and the `cpufreqctl` tool is based on the version number which remains unchanged in non-releases and therefore does not trigger the update procedure.

stefangweichinger, 2021-05-20:
@fin-ger Oh, I see, thanks. I will maybe re-install later. For now things seem to work .. and I am lazy/busy right now ;-)

bcclsn, 2021-05-21:
I've installed this release, it seems to work fine except the icon and the current frequency in the top bar, which are not displayed... as you can see in the attached image.
[image](https://postimg.cc/CB5DYJNm)
furthermore, a generic profile is loaded at startup instead of the one set for battery use. auto switch is enabled.

I'm on Archlinux with gnome 40.1

fin-ger, 2021-05-21:
@bcclsn The "Current Frequency" seems to be locked at 800 MHz, so something is absolutely not running smoothly here. This generally means that communication with the backend fails at a very early stage in initialization. Have you uninstalled and installed the tool and polkit rule after switching to the branch?

bcclsn, 2021-05-21:
mmm yes, you ar right...
i don't know if it is relevant from or not, but from cli the frequency changes correctly.

yes, I followed the whole procedure

fin-ger, 2021-05-21:
The cli received a major update, so it is important you have the new version installed otherwise it won't work. The development version isn't able to check for compatibility of the cli and the extension code as this is solely based on the version number.

Can you provide the the output of `journalctl` relevant to cpupower/gnome-shell?

bcclsn, 2021-05-23:
seems there is an error with metadata
`
mag 23 15:35:37 hostname gnome-shell[765]: JS ERROR: Could not load extension cpupower@mko-sl.de-backup: Error: Missing metadata.json`

bcclsn, 2021-05-23:
only cpupower
`mag 23 15:56:58 dreamspark pkexec[12908]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json min set 18]
                                               powerActions@/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:136:33
                                               enable@/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:112:14
                                               enableIndicator@/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:48:23
                                               enable/<@/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/extension.js:88:13
                                               spawnProcessCheckExitCode/<@/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/utils.js:63:21
mag 23 15:56:58 dreamspark pkexec[12910]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json turbo set off]
mag 23 15:56:58 dreamspark pkexec[12909]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json max set 18]
mag 23 15:56:58 dreamspark pkexec[12913]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json info frequencies]
mag 23 15:57:00 dreamspark pkexec[12948]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json min set 18]
mag 23 15:57:00 dreamspark pkexec[12949]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json max set 18]
mag 23 15:57:06 dreamspark pkexec[12967]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json min set 50]
mag 23 15:57:06 dreamspark pkexec[12968]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json max set 100]
mag 23 15:57:06 dreamspark pkexec[12971]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json turbo set on]
mag 23 15:57:06 dreamspark pkexec[12976]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json min set 50]
mag 23 15:57:06 dreamspark pkexec[12979]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json max set 100]
`
gnome-shell
`mag 23 15:55:13 dreamspark systemd[698]: Started Application launched by gnome-shell.
mag 23 15:55:21 dreamspark gnome-shell[8390]: meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_add: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_remove: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: JS ERROR: TypeError: adjustment is null
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_add: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_add: assertion 'display' failed
mag 23 15:56:40 dreamspark gnome-shell[8390]: meta_later_add: assertion 'display' failed
[... cut ...]
mag 23 15:57:06 dreamspark pkexec[12979]: bcclsn: Executing command [USER=root] [TTY=unknown] [CWD=/home/bcclsn/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-bcclsn --backend intel_pstate --format json max set 100]
mag 23 15:57:08 dreamspark systemd[12642]: Started Application launched by gnome-shell.
mag 23 15:57:14 dreamspark dbus-daemon[12658]: [session uid=1000 pid=12658] Activating service name='org.gnome.Nautilus' requested by ':1.11' (uid=1000 pid=12699 comm="/usr/bin/gnome-shell ")
mag 23 15:57:16 dreamspark gnome-shell[12699]: meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed`

fin-ger, 2021-05-23:
@bcclsn ah perfect 👍 I'll get to this asap, looks like I messed up the profile management.

fin-ger, 2021-06-03:
@bcclsn I reworked the initialization procedure. The issue arises when Auto-Switch is enabled and the profile is switched to the configured one before the menu was completely created (due to async stuff in the code). It's quite hard to test for me in VM's as the AC events cannot easily be triggered (or at least I don't know how). Can you test the latest commit of the `feature/modular-cpufreqctl-backends` branch and see if the issue is resolved?

bcclsn, 2021-06-03:
@fin-ger hi thanks for all! 
it seems to work but not very well ... every now and then it does not switch the profile automatically, I don't know if it is slow to do it or if it is blocked. it doesn't seem to change profile automatically when I manually set one. what test can i do to help you understand?

fin-ger, 2021-06-04:
Hmm, I'll see if I can get this reproduced on real hardware. I get back to you as soon as I have some progress on this.

fin-ger, 2021-06-05:
@bcclsn I could reproduce this issue (I think) on my X260 laptop.

The Auto-Switch works fine when already logged in. Unplugging and plugging the AC-adapter in an active Gnome session will switch the profiles reliably.

However, when logging in (the initial setup after successful login) the profile is not reliably switched. The profile is only applied partially (e.g. only the turbo boost or minimum frequency setting is applied). Is this the behavior you observed?

fin-ger, 2021-06-05:
@bcclsn I think I fixed this with the latest commit (7d93fe9).

@martin31821 @mastercaution The issues are all caused by the callback-based async-io in glib which is a pain to use in JS. I have not investigated if we can safely switch to a Promise based async-await. I fear that old Gnome platforms do not support this. As an example, this is how I fixed the aforementioned race condition:

```diff
     applyProfile(profile) {
         this.minVal = profile.MinimumFrequency;
-        this.updateMin();
-
-        this.maxVal = profile.MaximumFrequency;
-        this.updateMax();
-
-        this.isTurboBoostActive = profile.TurboBoost;
-        this.updateTurbo();
-
-        this.updateUi();
+        this.updateMin(() => {
+            this.maxVal = profile.MaximumFrequency;
+            this.updateMax(() => {
+                this.isTurboBoostActive = profile.TurboBoost;
+                this.updateTurbo(() => {
+                    this.updateUi();
+                });
+            });
+        });
     }
```

This is really ugly and with more lines of code, the nesting gets worse and worse.

Either way, porting the extension to proper async-await is out-of-scope for PR #151 and should be implemented separately (preferably after #151 is merged :laughing:) and definitely not a requirement for Gnome 40 support which is urgent for now. However, for the sake of maintainability, we should discuss and investigate on async-await support in the different Gnome platforms after Gnome 40 support is finished.

bcclsn, 2021-06-12:
@fin-ger yes, that was the problem and it seems solved with the latest commit. good job!!!
I will test in the next few days :)

schaze, 2021-06-20:
If it helps for test coverage, I am running an Arch Linux installation with Gnome 40.2.0. I can compile the version in the feature branch _feature/modular-cpufreqctl-backends_, install it and it works just fine. I have a Dell Laptop with a Intel® Core™ i7-9750H CPU.

fin-ger, 2021-06-20:
@schaze ah perfect :+1: Can you confirm that the frequency setting resets when opening the preferences window? I think this bug should also appear in Gnome 40.

Otherwise, I need someone to review PR #151, although not all issues are fixed by now (see the PR for details). I especially need people to try to break the extensions security model and see if e.g. privilege escalation is possible. I reworked a lot of the backend and do not want to throw anything in production without at least a basic level of security inspection.

Thanks in advance for any help!

eleius, 2021-06-21:
I am also running Arch Linux with Gnome 40.2, but when I make & install the version in feature/modular-cpufreqctl-backends, there seems to be a visual problem with the frequency sliders. See [screenshots here](https://imgur.com/a/EEQy9Za) (first image: just opened the menu, second image: trying to reduce maximum frequency by dragging the slider, but it also appears on the min freq line)

fin-ger, 2021-06-21:
@eleius This is expected as the minimum frequency cannot be above the maximum frequency. The bars indicate the minimum and maximum boundaries.

Any idea on how to make this visually more clear?

eleius, 2021-06-21:
@fin-ger ah, sorry! I thought it was a visual glitch because the max freq boundary was missing but it was shown above (min freq). Maybe just add the boundary to both, as for the leftmost boundaries?

fin-ger, 2021-06-21:
@eleius Technically, the maximum frequency slider has no maximum boundary. The maximum frequency slider defines the maximum boundary of the minimim slider. The maximum frequency itself can be anything greater than the minimum frequency, therefore there is no maximum boundary.

But I do get your confusion. I just don't know how to visualize this better.

eleius, 2021-06-21:
@fin-ger Perhaps using a different color (e.g. light gray) for the other slider's boundary would make it clearer that it's just a reference and not the "active" marker? (I don't know if it's possible to use different colors)

fin-ger, 2021-06-21:
@eleius Yes, it is. We initially used a grey-tone, but it was close to invisible using the high-contrast theme of GNOME. We chose the blue one as it had best visibility using different themes. More on that topic in #142

schaze, 2021-06-21:
@fin-ger I am not sure if I am testing it wrong, but my frequency settings do not reset.

Testcase:
1. Open menu, change Frequency settings (either using a profile or the sliders directly)
2. Close Menu, reopen menu and open preferences or directly open preferences in opened menu after frequency change
3. Close Preference window
4. Open menu, check frequency settings --> still the same

Not sure about PR #151 and security, this is not really my level of expertise. I will try and run the test series as described in the PR an report back.

ndm767, 2021-07-10:
I was having an issue on Ubuntu 21.04 with GNOME 40 on X11 where when I installed the extension, it would say Error: UNKNOWN in the drop-down menu. What seems to be causing this is that there are 2 installations of `cpufreqctl` installed, one at `/usr/bin/cpufreqctl` and one at `/usr/local/bin/cpufreqctl-[user]` where user is my username. The second one doesn't seem to support the `--backend` and `--format` flags. The way I fixed this was by doing `sudo mv /usr/local/bin/cpufreqctl-[user] /usr/local/bin/cpufreqctl-[user]-old` and then doing `ln -s /usr/bin/cpufreqctl /usr/local/bin/cpufreqctl-[user]` and then logging out and in after which it seems to work.

fin-ger, 2021-07-11:
@ndm767 Please see #156 for explanation on why this is not an issue.

sharkguto, 2021-07-29:
i am getting the following error too

The settings of extension cpupower@mko-sl.de had an error:
```
Gtk.BuilderError: /home/gustavo/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:88:1 Invalid object type 'GtkButtonBox'
```

Stack trace:
```
CPUPowerPreferences@/home/gustavo/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:53:22
buildPrefsWidget@/home/gustavo/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

sharkguto, 2021-07-29:
> i am getting the following error too
> 
> The settings of extension [cpupower@mko-sl.de](mailto:cpupower@mko-sl.de) had an error:
> 
> ```
> Gtk.BuilderError: /home/gustavo/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:88:1 Invalid object type 'GtkButtonBox'
> ```
> 
> Stack trace:
> 
> ```
> CPUPowerPreferences@/home/gustavo/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:53:22
> buildPrefsWidget@/home/gustavo/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
> _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
> OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
> asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
> run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
> main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
> run@resource:///org/gnome/gjs/modules/script/package.js:206:19
> start@resource:///org/gnome/gjs/modules/script/package.js:190:8
> @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
> ```

I am using gnome 40 in manjaro linux

fin-ger, 2021-07-29:
@sharkguto Please read the thread before posting to prevent unnecessary noise: https://github.com/martin31821/cpupower/issues/145#issuecomment-844933005

evgnor86, 2021-09-19:
Settings don't work on Gnome 40

fin-ger, 2021-09-27:
Will be fixed in next update

Neo-29, 2021-09-27:
What about gnome 41? broken too

fin-ger, 2021-09-27:
I don't know, I haven't tested it, yet. Have you tested the current master branch with a Gnome 41?

Neo-29, 2021-09-27:
> I don't know, I haven't tested it, yet. Have you tested the current master branch with a Gnome 41?

in fact it works. I rebuilt from AUR

Unfortunately, I have a leak on CPU usage about 1% . sure it's the extension, I tested witout.

May I open an issue?

Neo-29, 2021-09-27:
> Can you disable `Show current frequency` in the settings and check if the CPU usage is still increased?

Yep...that solves the issue

But I like to see the frequency

fin-ger, 2021-09-27:
Yes, but we need the CPU to calculate the frequency and display it in the top bar :sweat_smile: Although, we increased the complexity for how the frequency is fetched. This has something to do with the new abstraction layer supporting AMD Ryzen and Intel Core CPUs. We might improve on that in the future, but for now I would like to leave it as is, to not further delay the Gnome 40 release.

Neo-29, 2021-09-27:
> Yes, but we need the CPU to calculate the frequency and display it in the top bar 😅 Although, we increased the complexity for how the frequency is fetched. This has something to do with the new abstraction layer supporting AMD Ryzen and Intel Core CPUs. We might improve on that in the future, but for now I would like to leave it as is, to not further delay the Gnome 40 release.

I get it... No pb.
It was not the case before... I just noticed that, I just wanted to let you know the issue 😊

fin-ger, 2021-09-27:
I really appreciate it! It's really hard to test for all the different systems out there!

May I ask how many CPU cores your CPU has? This might have something to do with the increase.

Also, I'll create a tracking issue for this, so we can track progress on performance improvements for frequency fetching.

[4 more replies]