deinstapel/cpupower #148: Ryzen

Plateaus, 2021-02-20:
# Creating a new issue
please add amd support, I have this article:
https://sleeplessbeastie.eu/2015/11/09/how-to-set-cpu-governor-at-boot/

but still a gnome addon would be awesome!

fin-ger, 2021-02-21:
I might have a look at this in April along with the update to Gnome 40. See #145

fin-ger, 2021-02-21:
I reopen to use this as a tracking issue for AMD Ryzen support :)

fin-ger, 2021-05-02:
I started working on support for modular backends in the `cpufreqctl` script. I'm currently figuring out how to model the discrete scaling steps of AMD Ryzen CPUs in this extension. E.g. on my CPU 3 frequencies are available, in contrast to the seamless frequency scaling on intel_pstate (although these are also discrete, we're just not told which frequencies are available but it seems to be more than on Ryzen).

```
$ cat scaling_available_frequencies 
3600000 2800000 2200000
```

fin-ger, 2021-05-13:
Oh, I just realized that I haven't commented here, yet. The implementation for AMD Ryzen support is finished and available on the `feature/modular-cpufreqctl-backends` branch. It will make it to the next release and land in the master branch once #142 is done and Gnome 40 support (#145) is properly tested.

fin-ger, 2021-05-17:
Part of #151

fin-ger, 2021-09-27:
Will be included in next release