deinstapel/cpupower #149: Popover resizes depending on CPU frequence

amivaleo, 2021-04-09:
Hi!

I think the best way to explain the issue is to share this short video I made: https://streamable.com/mes6w8

I'm quite sure that resizing effect depends on the localization. And yet...

I hope there can be a simple fix for this. :)

fin-ger, 2021-05-13:
Oh, almost forgot about this one! I hope I don't forget to fix this for the next release... I'll get to this once all the other work regarding Gnome 40, the new slider element, AMD Ryzen support, and the new translations are done.

amivaleo, 2021-05-14:
> Oh, almost forgot about this one! I hope I don't forget to fix this for the next release... I'll get to this once all the other work regarding Gnome 40, the new slider element, AMD Ryzen support, and the new translations are done.

No problem. :)

I can't help with the coding, but I can be a very useful reminder.

commit ef06475e (2021-09-27): Fix #149 and #140