deinstapel/cpupower #153: [documentation request] auto-switch and turbo mode

juancrescente, 2021-06-15:
Hi!

I just want to know how the auto switch function works and what does the turbo mode exactly does.

Also I noticed that the selected profile is not selected after a while which could be related to the auto switch(?).

This extension is awesome

juancrescente, 2021-06-15:
it is not related to the auto switch since even if it's disabled, it is unselected

I have a fresh install of  Ubuntu 20.04.2 LTS

fin-ger, 2021-06-17:
Hmm, interesting... Maybe this is related to suspend or similar. We have issues restoring the settings after a suspend (although I'm currently not sure whether this is a kernel issue or it is a feature that the frequency scaling resets after a suspend).

Can you observe this behavior a little more and see if you can find any event which resets the frequency scaling settings?

fin-ger, 2021-06-17:
> I just want to know how the auto switch function works and what does the turbo mode exactly does.

Sorry, I totally missed answering your question!

So, Auto-Switch will automatically select a profile when you plug-in or plug-out your AC adapter on a laptop. If you are running this on a PC, Auto-Switch is mostly useless to you.

Turbo Boost is a feature of modern CPU's where the frequency of individual cores is boosted way above the base frequency (e.g. from 3GHz to 4GHz) to do fast calculations for a very short amount of time. Doing this for longer than a couple of minutes/seconds (depending on the frequency - 5GHz are only achieved for a fraction of a second) will result in overheating, therefore it is called a boost. It is a short boost for your CPU to do calculations quickly and then fall back to a lower frequency. This is most useful for e.g. smooth animations, fast launches of an application, etc. This feature has different names among the CPU vendors, we just use the intel name for it. AMD calls it Precision Boost and XFR (eXtended Frequency Range). Some also refer to it as `automatic overclocking`.

As some cooling systems especially on laptops are very bad in handling these boosts, we introduced the possibility to deactivate this on your machine. This will result in cooler CPU temperatures but also slower animations, applications startups, etc.

juancrescente, 2021-06-17:
thanks for the description!

juancrescente, 2021-06-17:
@fin-ger just fyi turbo boost is enabled automatically when I turn on auto switch, according to your explanation it probably shouldn't

closed 2021-06-17 by juancrescente

fin-ger, 2021-06-17:
Actually, it should be enabled by default as the CPU vendors aim for the highest performance and not for the lowest temperatures. So the default setting is enabled turbo boost, although not all users would prefer this.