deinstapel/cpupower #156: GNOME 40 unkown error on Xorg

ndm767, 2021-07-11:
CPU - Intel i7-11800H
Linux kernel version - 5.11.0-22-generic on Ubuntu 21.04
GNOME version - 40.2.0

I apologize that I did not think to save the `journalctl -xb` output before I fixed the problem.
I already mentioned this on the main GNOME 40 thread (#145) but I thought I might make an issue for it.

The extension worked fine on GNOME 3.38, however when I updated to GNOME 40 and installed the modular-freqctl-backends branch by running `sudo make install install-tool`, I received an error in the popup window that said Error: Unknown.

Upon reading the `journalctl -xb` output, it seems that the program is trying to run an executable at `/usr/local/bin/cpufreqctl-[user]` where `[user]` is my username. The main cpufreqctl installation is at `/usr/bin/cpufreqctl`. The one in `/usr/local/bin` was giving an error on the `--backend` and `--format` flags, saying that it does not know those flags. 
The way I fixed it for myself was moving the `/usr/local/bin/cpufreqctl-[user]` to `/usr/local/bin/cpufreqctl-[user]-old` and then creating a symlink from `/usr/bin/cpufreqctl` to `/usr/local/bin/cpufreqctl-[user]`. After logging out and logging back in, everything works great.

fin-ger, 2021-07-11:
Yes, this is intended. `sudo make install install-tool` does a system-wide install as stated in the [README](https://github.com/martin31821/cpupower#installing-for-all-users). However, your previous installation was not system-wide. I guess your previous installation was a user installation from the Gnome Extensions website. To solve this, I suggest uninstalling the system-wide installation with `sudo make uninstall uninstall-tool` and reinstall the Gnome 40 version as a normal user-installation: `make install PREFIX=/home/username/.local`. This only works if your user can issue admin commands via policykit which should be the case if you were able to successfully install from the Gnome Extensions website.

Please don't hesitate to ask if you encounter any further issues!

closed 2021-07-18 by ndm767

fin-ger, 2021-12-07:
Depending on your linux distribution you are now able to install this extension via your package manager. Consult the readme for detailed instructions.