deinstapel/cpupower #157: [Gnome 40/Fedora 34] error occurred after upgraded

arobot123, 2021-07-19:
# Creating a new issue

* Check if another issue with the same problem exists
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
  * policykit working and rule working
* Check if the cpufreqctl utility works correctly
* Please include the following information (only for bugreports):
  * CPU Model
  * Linux kernel version
  * policykit version
  * Gnome shell version

 % gnome-shell --version
GNOME Shell 40.3

  * journalctl -xb (maybe pastebin/filtered)

Plugin Errors Self Reported:

Gtk.BuilderError: /home/angusrobot/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:88:1 Invalid object type 'GtkButtonBox'

Stack trace:
  CPUPowerPreferences@/home/angusrobot/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:53:22
  buildPrefsWidget@/home/angusrobot/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:206:19
  start@resource:///org/gnome/gjs/modules/script/package.js:190:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

fin-ger, 2021-07-19:
Gnome 40 is not officially supported, see #151 for reference.

closed 2021-07-19 by fin-ger

peter-kehl, 2021-11-29:
Related/duplicate of #174.

fin-ger, 2021-12-03:
Just letting you know that I prepared a copr for Fedora and other RPM based distributions: https://copr.fedorainfracloud.org/coprs/fin_ger/cpupower/packages/

I will add this to the readme once I've finished the PPA setup and probably an update to the Arch AUR package.

peter-kehl, 2021-12-03:
Thank you. Especially looking forward to Arch AUR.

fin-ger, 2021-12-07:
Detailed installation instructions are now available in the README for many RPM based distributions.