deinstapel/cpupower #158: no "preferences"

m4jki, 2021-08-31:
The settings of extension cpupower@mko-sl.de had an error:
```
Gtk.BuilderError: /home/mike/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:88:1 Invalid object type 'GtkButtonBox'
```

Stack trace:
```
CPUPowerPreferences@/home/mike/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:53:22
buildPrefsWidget@/home/mike/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

fin-ger, 2021-08-31:
Gnome 40 is not yet supported in the latest release. Please refer to #151 and #145.

closed 2021-08-31 by fin-ger

crysman, 2022-03-27:
I've got the same issue on Pop OS 21.10 (Gnome 40.5) - no preferences (upon clicking on gears icon, no window popup), other extensions work fine.
**According to README, GNOME 40 should be supported** - so, what is wrong?
PS: I used to have this extension installed manually before once it had been possible via `Tweaks` - but some time ago, all extensions instaled this way got disabled, so I used the README way (PPA)...
Any fix, please @fin-ger ?

fin-ger, 2022-04-01:
Have you uninstalled the previously installed version?

crysman, 2022-04-01:
I do not know how, since extensions section has disappeared in Tweaks (GNOME Tweaks 40), have a look:
[image]

So.. how do I do it, please?

fin-ger, 2022-04-01:
Have you installed it with the Makefile?

crysman, 2022-04-01:
Nope, classic apt:
```
❱ dpkg -l | grep cpupower
ii  gnome-shell-extension-cpupower            10.0.1-1~impish1                                             all          Gnome-Shell Extension for controlling the frequency setting of your CPU
```

fin-ger, 2022-04-01:
Also, I can't remember if we installed polkit stuff under `/usr/local/share/polkit-1`. Better also look there. I could check the old sources but I think it's faster to just check if there are any files.

crysman, 2022-04-01:
third path there is no such a thing:
```
❱ l /usr/share/polkit-1/rules.d/
20-gnome-initial-setup.rules  org.freedesktop.Flatpak.rules
60-network-manager.rules      org.freedesktop.fwupd.rules
geoclue-2.0.rules             org.freedesktop.packagekit.rules
gnome-control-center.rules    org.gtk.vfs.file-operations.rules
org.freedesktop.bolt.rules    systemd-networkd.rules
```

but the first two I have deleted.

in local I have only .bak.disabled (i renamed so myself)...
```
❱ l ~/.local/share/gnome-shell/extension
extensions.bak.disabled/ extension-updates/
```

so it should be disabled.
now rebooting (just to be sure)

crysman, 2022-04-01:
no :( It's not helped, still no preferences... Are there any logs of this?

fin-ger, 2022-04-01:
`journalctl /usr/bin/gnome-shell`

crysman, 2022-04-01:
hmm, I'm not smart of it :( are you?
```
❱ sudo journalctl /usr/bin/gnome-shell | grep -Ei 'cpu|power' | tail
dub 01 17:47:00 SPIN5 gnome-shell[2828]: gnome-shell-extension-system76-power: power profile was set: 'Balanced'
dub 01 17:47:02 SPIN5 gnome-shell[2828]: Power state changed: fully charged
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
dub 01 18:24:16 SPIN5 gnome-shell[2477]: gnome-shell-extension-system76-power: power profile was set: 'Balanced'
dub 01 18:25:08 SPIN5 gnome-shell[2753]: gnome-shell-extension-system76-power: power profile was set: 'Balanced'
dub 01 18:25:10 SPIN5 gnome-shell[2753]: Power state changed: fully charged
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
```

could it be somehow in conflict with that `gnome-shell-extension-system76-power` ?

fin-ger, 2022-04-01:
Hmm, I don't know. The error occurs in this line:

```js
Util.trySpawnCommandLine(`${EXTENSIONDIR}/src/prefs40/main.js`);
```

Can you run this:

```
$ journalctl -b | grep -C 10 -Ei 'cpupower'
```

crysman, 2022-04-01:
here its is:

```
myuser@SPIN5 ~ 06:41:49 
❱ journalctl -b | grep -C 10 -Ei 'cpupower'
Hint: You are currently not seeing messages from other users and the system.
      Users in groups 'adm', 'systemd-journal' can see all messages.
      Pass -q to turn off this notice.
dub 01 18:25:11 SPIN5 systemd[1681]: app-gnome-com.system76.PopUpgrade.Notify-2979.scope: Deactivated successfully.
dub 01 18:25:11 SPIN5 gsd-color[2911]: failed to connect to device: Failed to connect to missing device /org/freedesktop/ColorManager/devices/xrandr_BOE_gdm_110
dub 01 18:25:11 SPIN5 dbus-daemon[1701]: [session uid=1000 pid=1701] Activating service name='org.gnome.ArchiveManager1' requested by ':1.86' (uid=1000 pid=4048 comm="gjs /usr/share/gnome-shell/extensions/ding@rasters")
dub 01 18:25:11 SPIN5 systemd[1681]: Started Application launched by gnome-session-binary.
dub 01 18:25:11 SPIN5 slimbookbattery-autostart.desktop[4070]: python3: can't open file '/usr/share/slimbookbattery/src/slimbookbatteryindicator.py': [Errno 2] No such file or directory
dub 01 18:25:11 SPIN5 systemd[1681]: app-gnome-slimbookbattery\x2dautostart-4070.scope: Deactivated successfully.
dub 01 18:25:11 SPIN5 dbus-daemon[1701]: [session uid=1000 pid=1701] Successfully activated service 'org.gnome.ArchiveManager1'
dub 01 18:25:15 SPIN5 gnome-shell[2753]: JS ERROR: GLib.SpawnError: Permission denied
                                         trySpawn@resource:///org/gnome/shell/misc/util.js:128:19
                                         trySpawnCommandLine@resource:///org/gnome/shell/misc/util.js:160:13
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
                                         activate@resource:///org/gnome/shell/ui/popupMenu.js:191:14
                                         vfunc_button_release_event@resource:///org/gnome/shell/ui/popupMenu.js:138:14
dub 01 18:25:17 SPIN5 systemd[1681]: tracker-extract-3.service: Deactivated successfully.
dub 01 18:25:20 SPIN5 gnome-shell[2753]: JS ERROR: GLib.SpawnError: Permission denied
                                         trySpawn@resource:///org/gnome/shell/misc/util.js:128:19
                                         trySpawnCommandLine@resource:///org/gnome/shell/misc/util.js:160:13
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
                                         activate@resource:///org/gnome/shell/ui/popupMenu.js:191:14
                                         vfunc_button_release_event@resource:///org/gnome/shell/ui/popupMenu.js:138:14
dub 01 18:25:21 SPIN5 systemd[1681]: run-user-110-gvfs.mount: Deactivated successfully.
dub 01 18:25:22 SPIN5 systemd[1681]: run-user-110.mount: Deactivated successfully.
dub 01 18:25:22 SPIN5 gnome-shell[2753]: JS ERROR: GLib.SpawnError: Permission denied
                                         trySpawn@resource:///org/gnome/shell/misc/util.js:128:19
                                         trySpawnCommandLine@resource:///org/gnome/shell/misc/util.js:160:13
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
                                         activate@resource:///org/gnome/shell/ui/popupMenu.js:191:14
                                         vfunc_button_release_event@resource:///org/gnome/shell/ui/popupMenu.js:138:14
dub 01 18:25:24 SPIN5 systemd[1681]: Started Application launched by gnome-shell.
dub 01 18:25:24 SPIN5 gnome-keyring-daemon[1702]: asked to register item /org/freedesktop/secrets/collection/login/1, but it's already registered
dub 01 18:25:24 SPIN5 gnome-shell[4305]: libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
dub 01 18:25:24 SPIN5 gnome-shell[4305]: [4334:4334:0401/182524.576534:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
dub 01 18:25:24 SPIN5 gnome-keyring-daemon[1702]: asked to register item /org/freedesktop/secrets/collection/login/1, but it's already registered
dub 01 18:25:25 SPIN5 gnome-shell[4305]: [4306:4306:0401/182525.073170:ERROR:sharing_service.cc(218)] Device registration failed with fatal error
dub 01 18:25:39 SPIN5 gnome-shell[2753]: Window manager warning: Overwriting existing binding of keysym 6f with keysym 6f (keycode 20).
dub 01 18:26:08 SPIN5 systemd[1681]: Started Application launched by gnome-session-binary.
dub 01 18:26:09 SPIN5 io.elementary.a[5310]: failed to set proxy: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._pk_2dengine_2derror_2dquark.Code3: setting the proxy failed: failed to get the uid
dub 01 18:26:09 SPIN5 io.elementary.a[5310]: PackageKitBackend.vala:187: Unable to set proxy for PackageKit to use: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._pk_2dengine_2derror_2dquark.Code3: setting the proxy failed: failed to get the uid
dub 01 18:26:16 SPIN5 gnome-shell[2753]: JS ERROR: GLib.SpawnError: Permission denied
                                         trySpawn@resource:///org/gnome/shell/misc/util.js:128:19
                                         trySpawnCommandLine@resource:///org/gnome/shell/misc/util.js:160:13
                                         onPreferencesActivate@/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js:566:18
[... cut ...] bexec/xdg-desktop-portal ")
dub 01 18:35:48 SPIN5 systemd[1681]: Starting flatpak document portal service...
dub 01 18:35:48 SPIN5 gnome-keyring-daemon[1702]: asked to register item /org/freedesktop/secrets/collection/login/1, but it's already registered
dub 01 18:35:48 SPIN5 dbus-daemon[1701]: [session uid=1000 pid=1701] Successfully activated service 'org.freedesktop.portal.Documents'
dub 01 18:35:48 SPIN5 systemd[1681]: Started flatpak document portal service.
myuser@SPIN5 ~ 06:43:12 
❱ 

```

fin-ger, 2022-04-01:
```
dub 01 18:25:15 SPIN5 gnome-shell[2753]: JS ERROR: GLib.SpawnError: Permission denied
```

Interesting. You are not to allowed to launch the preferences dialog. Can you run

```
$ cat /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js
```

and

```
$ ls -l /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js
```

crysman, 2022-04-01:
```
myuser@SPIN5 ~ 06:49:45 
❱ cat /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js && ls -l /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js
#! /usr/bin/env gjs

const Gio = imports.gi.Gio;

let scriptdir = Gio.File.new_for_path(imports.system.programPath).get_parent();
imports.searchPath.unshift(scriptdir.get_path());
imports.searchPath.unshift(scriptdir.get_parent().get_parent().get_path());

imports.gi.versions.Gtk = "3.0";
const Gtk = imports.gi.Gtk;
const prefs = imports.prefs;

Gtk.init(null);
prefs.init();

let win = new Gtk.Window();
win.connect("delete-event", () => {
    Gtk.main_quit();
});
let headerbar = new Gtk.HeaderBar();
headerbar.set_show_close_button(true);
headerbar.title = "CPU Power Manager";
win.set_titlebar(headerbar);
let mainWidget = prefs.buildPrefsWidget();
win.add(mainWidget);
win.show_all();

Gtk.main();
-rw-r--r-- 1 root root 705 pro  8 19:34 /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js
myuser@SPIN5 ~ 06:50:10 
❱ 

```

fin-ger, 2022-04-01:
Huh... the executable bit is missing.

Do

```
$ sudo chmod +x /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js
```

This should be done by apt. Idk why this is not working. But I currently don't have the time to dig into the packaging bug.

crysman, 2022-04-01:
yes, fixed, thanks a lot!

fin-ger, 2022-04-01:
Nah, this **can** confuse your package manager and will be broken when some apt script revisits the cpupower package.

crysman, 2022-04-01:
hmm, I see... I will probably wait for patch then. Thanks for noticing me.