deinstapel/cpupower #172: Switch to other profile impossible

jeed2424, 2021-11-08:
cpu: i7-6500u
gnome version: 3.36.8
Linux Kernel: 5.4.0-58-generic

When I select a profile other than the first one on my list, the profile gets switched back to the first one within seconds. 

*See video for better explanation.

https://user-images.githubusercontent.com/50756835/140802629-c75bd991-33f9-4b92-81ba-9237ae856fb3.mp4

pdragicevich, 2021-11-27:
I had this issue as well, found out it was because I had CPU power features disabled in the BIOS. (Writes to /sys/devices/system/cpu/intel_pstate/no_turbo were faililng). Still does some weird things such as not setting minimum CPU frequency below 45%, probably it's the same thing, haven't bothered to explore. It will only "stick" on a profile if the settings will stay at those values.

fin-ger, 2021-11-28:
@pdragicevich This might have sth to do with the minimum supported hardware frequency. The version 10.0.0 of this extension includes a great UI rework by @mastercaution which visualizes this way better than previous versions. Unfortunately, this version is not allowed to be published on the GNOME extensions website unless we drop support for old GNOME versions and do a complete rewrite of our GNOME 40 support. As this has been communicated to us after we finishes all testing and already made the final release, it is unlikely that this version will ever land on the GNOME extensions website. Also, we currently have no time (and especially motivation) for a full rewrite. Therfore you have to manually install the update as described in the README until I (or someone else) gets rpm and deb packages ready (ideally distributed via copr and ppa). I successfully packaged the extension as rpm and deb last week, but haven't had time to integrate it with build services, yet. But if you're patient, you might be able to update via your distributions package manager in the next couple of months.

Keltere, 2022-05-30:
Same here with ryzen 3900xt of fedora 36. Other than that the stock clock is wrong, it should be 3.8GHz instead of 3.6GHz 

https://user-images.githubusercontent.com/8084245/171023057-6c7ab157-7523-42f9-8fa7-94febd5fe450.mp4

mastercaution, 2022-05-30:
@Keltere Looks like something else is resetting your frequency preferences (driver or other power tools)

Keltere, 2022-05-31:
> @Keltere Looks like something else is resetting your frequency preferences (driver or other power tools)
```
cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver 
acpi-cpufreq
```

Keltere, 2022-05-31:
> 
What about amd_pstate is that a viable solution?

fin-ger, 2022-05-31:
I don't know. But feel free to add a new backend for it:

https://github.com/deinstapel/cpupower/blob/master/tool/cpufreqctl#L301

You have to implement the following functions:

```
<name>_init
<name>_supported
<name>_reset
<name>_turbo_get
<name>_turbo_set
<name>_min_get
<name>_min_set
<name>_max_get
<name>_max_set
<name>_info_frequencies
<name>_info_current
```

and integrate it into `backends_list`, `backend_select`, `help_main`, and `main`.

Looking forward to a PR implementing this!