deinstapel/cpupower #173: AMD Ryzen

pparik, 2021-11-22:
Hi,
the readme states that this extension supports non-Intel processors (specifically the AMD Ryzen is mentioned), but it does not work on my AMD Ryzen 3950. Am I missing something?
Regards
P

```
$ uname -a
Linux 5.14.18-100.fc33.x86_64 #1 SMP Fri Nov 12 17:38:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
$ gnome-shell --version
GNOME Shell 3.38.5
$ cpufreqctl
CPU driver: acpi-cpufreq
Governors: conservative ondemand userspace powersave performance schedutil
Frequencies: 3500000 2800000 2200000
```

fin-ger, 2021-11-22:
Version 10.0.0 supports AMD Ryzen. However, version 10.0.0 is not allowed to be published on the GNOME extensions website. You have to install it from the release page of this repo or manually as described in the readme

pparik, 2021-11-22:
Thank you, it works!

closed 2021-11-22 by pparik