deinstapel/cpupower #174: Preferences fail on Manjaro GNOME 41.1

peter-kehl, 2021-11-29:
Existing working installation of CPUPower on Manjaro Linux x64. It still works (including switching between existing profiles, as it used to), but Preferences now fail with the below error. Same error with both the current stable kernel and release candidate kernel:
```
X11, GNOME 41.1
5.15.2-2-MANJARO #1 SMP PREEMPT Sat Nov 13 19:25:38 UTC 2021 x86_64 GNU/Linux
5.16.0-1-MANJARO #1 SMP PREEMPT Mon Nov 15 18:57:36 UTC 2021 x86_64 GNU/Linux

Gtk.BuilderError: /home/pkehl/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/data/cpupower-preferences.glade:88:1 Invalid object type 'GtkButtonBox'

Stack trace:
  CPUPowerPreferences@/home/pkehl/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/preferences.js:53:22
  buildPrefsWidget@/home/pkehl/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/prefs.js:41:23
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:219:40
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:128:33
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:206:19
  start@resource:///org/gnome/gjs/modules/script/package.js:190:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

```

Unsure of `cpufreqctl` - running `cpufreqctl` says `command not found`. Running on HP Spectre, Intel i7-5600U.

fin-ger, 2021-11-29:
It looks like version 10.0.0 might fix it. Although 10.0.0 also only support GNOME 40, the GtkButtonBox error is Gtk4 related. However, we are not allowed to publish version 10.0.0 to the Gnome extension website by the Gnome extension administrators. I'm currently looking into creating rpm and deb packages and distributing it over Fedora coprs and Ubuntu ppa. This extension is already packaged in an AUR. You might be lucky to get it working by editing the metadata.json, but previous tests with GNOME 41 on Debian testing were unsuccessful for me.

Nevertheless, the motivation to still work on this is quite low as the last Release was shutdown by the GNOME admins with requirements of us dropping support for any release before GNOME 40 and doing a complete rewrite of our GNOME 40 support. We did not receive an answer to our response until today, so any publication of an update to the GNOME extensions page seems unlikely.

Sorry for the inconvenience...

peter-kehl, 2021-11-29:
Thank you for explaining @fin-ger.

Summary of the below: `GIT/master` works in GNOME 41.1 in that it does show the existing frequency profiles/configurations. However GNOME steps on its feet and uninstalls it or hides the controls from the desktop bar.

Good news: On my above Manjaro (stable kernel 5.15) GNOME 41.1 I've manually removed `'~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de` and installed `GIT/master` by running `make install PREFIX=/home/pkehl/.local` - all in the same GNOME session (without logging out). That made "Preferences" show the existing profiles, and all tabs in "Preferences" seemed to be working.

Bad news: However, after I logged out and logged in a few times, sometimes "Extensions" app and/or G. Chrome > showed "CPU Power Manager" as in a need of an update, or one of those two or GNOME itself removed CPU Power Manager: it didn't show up in "Extensions" neither in Chrome > https://extensions.gnome.org/local/ anymore, and there was no `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de` anymore.

After a few manual GIT-based re-installs, I've re-installed the (current) version from GNOME portal. Please let me know if you'd like me to test by installing manually later. Feel free to close this issue or put to another state, as you like.

fin-ger, 2021-12-03:
The AUR package seems rather outdated. @tonylambiris I think you are maintaining the AUR package, right? Do you have time to update the package to a newer version? I think the PKGBUILD might not work with the latest version. You can have a look at the [`deb`](https://github.com/deinstapel/cpupower/blob/master/dist/deb/gnome-shell-extension-cpupower/debian/rules) and [`rpm`](https://github.com/deinstapel/cpupower/blob/master/dist/rpm/SPECS/gnome-shell-extension-cpupower.spec) packaging in this repo. If you like you can also add your PKGBUILD to this repo by placing it in the `dist` folder.

fin-ger, 2021-12-03:
> G. Chrome > showed "CPU Power Manager" as in a need of an update, or one of those two or GNOME itself removed CPU Power Manager

Yes, this is a bit flaky when using a manually installed extension. You also always have to log out and back in again to make the extension appear. You can only ignore those update messages, they are meaningless. I hope this will get better with native packaging in your distro.

peter-kehl, 2021-12-03:
Thank you Fin and Tony. Looking forward to an update on Arch AUR.

fin-ger, 2021-12-07:
@martin31821 @jwuensche @johannwagner @bengsparks Does anyone have a working Arch and time to get a PKGBUILD done for the AUR?

peter-kehl, 2021-12-10:
I've commented at https://aur.archlinux.org/packages/gnome-shell-extension-cpupower-git/#comment-840128, hopefully it will notify the AUR maintainer.

fin-ger, 2021-12-10:
Thank you for following up on this! Please keep in mind that GNOME 41 is still not supported by this extension.

If you are running GNOME 40, you can now update via the GNOME extensions website. When disabling the version check of gnome shell extensions you might be able to use this extension.

I myself only have a Debian testing machine setup which runs GNOME 41 (in a virtual machine, my daily driver is a Fedora 34), so my tests currently only includes one distro. When I have some time I'll setup a Fedora 35 as well. But any testing on the new GNOME release will help shipping a compatible version.

peter-kehl, 2021-12-10:
After updating (in Chrome, from GNOME repository) and rebooting, in GNOME 41.2 (updated today)
[image]

The extension doesn't show up in my GNOME bar anymore. However, it reports no errors (when seen in Chrome, neither in "Extensions" app):
[image]
And when I click at the options (wheel) icon in "Extensions" app, it does show CPUPower's options well, including my existing power profiles:
[image]

Happy to report more, if you suggest file paths, or commands.

In the meantime, as a workaround for anyone on newer GNOME, may I suggest https://github.com/vagnum08/cpupower-gui. Not as intuitive, not integrated in GNOME bar, and editing profiles is possible only outside the application - but still useful. Manjaro/Arch has it on AUR.

fin-ger, 2021-12-11:
I think the extension is not showing up because GNOME is preventing it from being started in the first place (version incompatibility).

For GNOME 41 to run this extension although it is currently marked as incompatible you have to tell GNOME to skip the version check:

```
gsettings set org.gnome.shell disable-extension-version-validation true
```

Please make sure you really want this. This will not only allow this extension to run on unsupported platforms, but all incompatible GNOME shell extensions will be installable. This can greatly reduce system stability and may have unwanted side effects. To disable this behavior, just set the value to `false` in the above command.