deinstapel/cpupower #176: Upgraded the gnome extension on my PopOS!_21.04 and it doesn't work anymore

dodalovic, 2021-12-08:


fin-ger, 2021-12-08:
Could you please describe in more detail what's happening?

If you just updated and an error indicator is displayed next to the extension's activation toggle button, please log out of your Gnome session and back in again to make the extension work. This is a bug by GNOME itself and can't be solved by any extension. I'm sorry you are having trouble updating.

fin-ger, 2021-12-08:
@martin31821 Can you place a hint in the description of the GNOME extensions website?

marioaelius, 2021-12-08:
On Ubuntu 20.04 the same.

mahpah, 2021-12-08:
I got the same issue on ubuntu 20.04 (just update the extension 5mins ago)

kernel 5.11.0-41-generic
gnome 3.36.8
processor Intel® Core™ i7-10875H

marioaelius, 2021-12-08:
Maybe it has something to do with the Backend option from Preferences
Frequency scaling driver = Automatic (unavailable)

marioaelius, 2021-12-08:
Ubuntu 20.04.3 LTS
Kernel 5.4.0-59-generic
Gnome 3.36.8
CPU Intel Core i7-10875H

marioaelius, 2021-12-08:
Or maybe it has to do something with the admin rights because I tried to set it manually and I had to use sudo
cpufreqctl-mas max 30
/usr/local/bin/cpufreqctl-mas: line 203: /sys/devices/system/cpu/intel_pstate/max_perf_pct: Permission denied

fin-ger, 2021-12-08:
Can you please try uninstalling the cpufreqctl tool in the preferences section and reinstall it? I assume a log out did not help?

As a workaround for Ubuntu based installations: use the ppa described in the readme. Everything should work there.

Another thing: there exists an [issue template](https://github.com/deinstapel/cpupower/blob/master/.github/ISSUE_TEMPLATE.md) for this repository. I know that commenters on this issue don't see this template, but it would really help tackling down the problem if someone experiencing this issue would provide the information requested in the template. That's why it exists in the first place.

Abhigyan-Mishra, 2021-12-08:
I had the same issue on PopOS_21.04 , I tried uninstalling through the preferences menu and reinstalling and that worked for me

fin-ger, 2021-12-08:
Ah, thanks for letting me know. This means I screwed up the version checks for the reinstall dialog (yes there exists one, it should have shown up). Sorry for that. I tested this previously but as we were forced to re-release by the GNOME extensions admins this must have slipped through. I will prepare a fix, but since it's the GNOME extensions website, approval might be tomorrow or in 6 months. Nobody knows.

In the meantime I kindly ask you to use the Ubuntu PPA or Fedora Copr instead. The packages there use another installation method and are not affected. Sorry for the inconvenience.

fin-ger, 2021-12-08:
Stupid Github, don't close...

marioaelius, 2021-12-08:
I installed this extension from extensions.gnome.org a long time ago and until today I had no idea about the github repository nor the ppa.
I uninstalled and reinstalled it from the ppa and everything is ok.
Thank you.

petr-hybler, 2021-12-08:
same here ... after update 

Oh no, something went wrong
An internal error occurred: UNKNOWN

Ubuntu 20.04.3
Intel® Core™ i7-8565U CPU @ 1.80GHz × 8
Gnome 3.36.8

fin-ger, 2021-12-08:
@marioaelius yes, the ppa has been set up yesterday, so brand new. I went this route as I hope to get updates shipped faster and easier.

pparik, 2021-12-08:
Same here on Fedora 33. The extension stopped working after updating to 10.0.0.1, had to manually downgrade back to 9.0.4.

Linux 5.14.18-100.fc33.x86_64
Intel® Core™ i7-8550U CPU @ 1.80GHz

marioaelius, 2021-12-08:
I think the ppa route is great.
Thanks again :)

ooctavianus, 2021-12-08:
got here from the report error ui part :d , nice touch btw !
Thanks for the extension, works great!

fin-ger, 2021-12-08:
@pparik there is also a Copr for Fedora 33-35

mahpah, 2021-12-11:
Just trying out the ppa today (the logout part stop me to try it earlier). It work like a charm, and I can see some nice UI tweak. Thanks