deinstapel/cpupower #177: FREQUENCY CONTROL

farrukh2002, 2021-12-08:
I CANNOT CONTROL MY CPY FREQUENCY BELOW 11 PERCENT. 
HOW IT HAPPENNED?
I UPDATED THE CPU MONITOR AND IT JUST KEPT FAILING AND THEN I UNINSTALLED IT. REBOOTED THE PC AND TRIED REINSTALLING IT. IT DID INSTALL BUT MY MINIMUM FREQUENCY WAS STUCK AT 11 PERCENT{ WHICH WAS PREVIOUSLY SET BY ME}.
IN THE PREVIOOUS VERSION IT USE TO GO AS LOW AS 1 PERCENT.

farrukh2002, 2021-12-08:
[image]
SEE THIS SS

fin-ger, 2021-12-08:
First of all, please keep your voice down! There is no need to shout!

Second, there is an issue template you did not follow. It helps in identifying your issue.

I assume you have an Intel Core processor. It has a hardware limitation where you can't set the minimum frequency lower than a specific value (differs from CPU model to CPU model). Previously this hardware limitation has not been correctly reflected in the user interface. Now it is. Your CPU was never actually running at 1%, it was not correctly displayed by this extension in the past. This has led to confusion by users, therfore we changed it and are now displaying what is actually implemented by the CPU.

fin-ger, 2021-12-12:
No response, closing.

closed 2021-12-12 by fin-ger

farrukh2002, 2021-12-14:
BRO I WAS NOT SHOUTING. MY KEYBOARD IS ALWAYS ON CAPS MODE SINCE MY WORK REQUIRES IT TO BE AS IT IS. I AM SORRY IF YOU FOUND THIS AN AGGRESSIVE MESSAGE BUT I HAD NO INTENTIONS LIKE THIS.