deinstapel/cpupower #178: Fails on Ubuntu 20.04

dzink, 2021-12-08:
The latest version isn't working for me:

When I try to run one of the commands in bash that I see in journalctl, looks like some don't work: 
```
$sudo /usr/local/bin/cpufreqctl-daniel --backend automatic --format json info frequencies
usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
version 9.0.2
error: unknown action --backend automatic --format json info frequencies
```

## More info

I originally installed this via the gnome plugin for firefox. It has been working well up until the most recent update.

After the most recent update, the icon disappeared, so I found this repo and I ran the Ubuntu 18.04 instructions in the README.

I installed the new source and installed the application. When I went to run the command to install it on gnome, I got:

```
$gnome-shell-extension-tool -e cpupower@mko-sl.de
gnome-shell-extension-tool is deprecated, use gnome-extensions instead
```

Instead I tried gnome-extensions.

```
$gnome-extensions enable cpupower@mko-sl.de
```

No output. Seems to be standard for this program.

I also restarted my computer as instructed.

On restart, the extension no longer showed the frequency. On clicking on the icon, it shows the message:

```
Oh no! Something went wrong.
An internal error occurred: UNKNOWN
```

I've tried reinstalling as well as wiping the settings in dconf.

Policykit version:
```
$pkaction --version
pkaction version 0.105
```
Ubuntu 20.04.3 LTS
Gnome shell version 3.36.8
Intel® Core™ i7-8565U CPU @ 1.80GHz × 8
Linux 5.11.0-41-generic x86_64

Journalctl is filled with these:
```
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gjs[14857]: JS LOG: Failed to query supported frequency ranges from cpufreqctl, reason UNKNOWN! Assuming full range...
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[11118]: Failed to query supported frequency ranges from cpufreqctl, reason UNKNOWN! Assuming full range...
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17936]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17937]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17938]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17939]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[11118]: Failed to query supported frequency ranges from cpufreqctl, reason UNKNOWN! Assuming full range...
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17940]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17941]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17942]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17943]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17944]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th pkexec[17945]: daniel: Executing command [USER=root] [TTY=unknown] [CWD=/home/daniel/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-daniel --backend automatic --format json info frequencies]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17945]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[11118]: Failed to query supported frequency ranges from cpufreqctl, reason UNKNOWN! Assuming full range...
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th pkexec[17948]: daniel: Executing command [USER=root] [TTY=unknown] [CWD=/home/daniel/.local/share/gnome-shell/extensions/cpupower@mko-sl.de] [COMMAND=/usr/local/bin/cpufreqctl-daniel --backend automatic --format json info frequencies]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17948]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17951]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17952]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17953]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17954]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[11118]: Failed to query supported frequency ranges from cpufreqctl, reason UNKNOWN! Assuming full range...
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17955]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17956]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17957]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:14 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17958]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:24 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17964]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:24 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17965]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 08 16:47:24 daniel-ThinkPad-X1-Carbon-7th gnome-shell[17966]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]

```

fin-ger, 2021-12-08:
please see #176

fin-ger, 2021-12-08:
as a workaround use the ppa or uninstall and install again in the preferences of this extension.

fin-ger, 2021-12-08:
On a side note, thanks for the detailed report, I really appreciate that. This makes solving issues way easier. It's btw fixed in version 10.0.1 but I don't know when and if it will land on the GNOME extensions website.