deinstapel/cpupower #179: An internal error occured: UNKNOWN after upgrade on Dec 8th 2021

iqbalabd, 2021-12-08:
# Creating a new issue

Hi there,

"The extension was working fine but stopped working after the recent upgrade which I did yesterday. It only shows "Oh no! Something went wrong. An internal error occurred: UNKNOWN" on the status bar.

These are my setup:
```
$ grep -i pstate /boot/config-$(uname -r)
CONFIG_X86_INTEL_PSTATE=y
$ uname -a
Linux salvador 5.4.0-91-generic #102~18.04.1-Ubuntu SMP Thu Nov 11 14:46:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
$ gnome-shell --version
GNOME Shell 3.28.4
$ pkaction --version
pkaction version 0.105
```

If there is anything else you'd like to know, please inform me and I'll add it here.

iqbalabd, 2021-12-09:
I followed the instructions from the README file and installed from Ubuntu's PPA instead. At first it didn't work (nothing changed), so I went to the Gnome's extension site, uninstalled the extension from there, restarted and reinstalled from PPA again following the instructions on README. Now it is working as expected.

But the Gnome extension site now shows that the extension needs "updating". So for now I'm just ignoring that.

fin-ger, 2021-12-09:
Yes, you will have to ignore that.

Thank you for reporting this. It is fixed in version 10.0.1, but not published on the GNOME extensions website, yet.

See #176.

fin-ger, 2021-12-12:
The new version is now published on the Gnome extensions website. Is your issue fixed by this update?

iqbalabd, 2021-12-13:
Thanks. I have updated it from the Gnome extensions website and it looks like it's working as expected.

iqbalabd, 2021-12-13:
I tried updating it from the Gnome extension websites, but it shows an error afterwards. Now I'm still using the installed version which I got from `apt-get` like what I wrote above, and it still works well.