deinstapel/cpupower #181: An internal error occurred: UNKNOW

edsonschleiwabion, 2021-12-09:
# An internal error occurred: UNKNOW

After extension update, it stops to work :(, it seems that it is trying to use some invalid cepufreqctl parameter.

* Check if another issue with the same problem exists 
   * nothing else found
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
    * OK
  * policykit working and rule working
    * OK
* Check if the cpufreqctl utility works correctly
  * $ cpufreqctl --version
  *  version: 10.0.1

* Please include the following information (only for bugreports):
  * CPU Model - Model name:                      Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
  * Linux kernel version - 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
  * policykit version (how to check this ?)
  * Gnome shell version - 3.36.8
  * journalctl -xb (maybe pastebin/filtered)
----

Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32129]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32129]: version 9.0.4
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32129]: error: unknown action --backend automatic --format json turbo get
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32130]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32130]: version 9.0.4
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32130]: error: unknown action --backend automatic --format json min get
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32131]: usage: cpufreqctl [-h] [turbo {get,0,1}] [min {get,check,VALUE}] [max {get,VALUE}]
Dec 09 09:21:32 ws-ab-schlei.local.wabion.de gnome-shell[32131]: version 9.0.4

fin-ger, 2021-12-09:
> it seems that it is trying to use some invalid cepufreqctl parameter.

Yes, you are right. Many thanks for the detailed report! Please see #176.

As a workaround you can either uninstall the backend tool in the settings of this extension and install it again or use the PPA as described in the readme of this project.

edsonschleiwabion, 2021-12-09:
It is working now after removing it using the "https://extensions.gnome.org/local/" and installing it with apt.

closed 2021-12-09 by edsonschleiwabion