deinstapel/cpupower #183: Max frequency slider seems broken

hidai, 2021-12-23:
Maximum frequency slider doesn't work as expected.

* CPU Model : `11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz`
* Linux kernel version: `Linux dellxps 5.11.0-40-generic #44-Ubuntu SMP Wed Oct 20 16:16:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
* policykit version: `policykit-1    0.105-30ubuntu0.1 amd64`
* Gnome shell version: `gnome-shell    3.38.4-1ubuntu3~21.04.1 amd64`

[image]

fin-ger, 2021-12-23:
Hmm, maybe the pstate API for 11th gen Intel core processors changed. I will not have access to one in the near future to verify the support. I will need other people to do some testing and development for that platform.

I will consult the kernel docs and see if there is anything documented regarding this.

fin-ger, 2021-12-23:
To just solve your problem of a slow pc: uninstall this extension and power off your machine (no reboot, might not alwayswork). Then wait a couple of seconds and turn it back on. The CPU should now have forgotten all configuration of cpupower.

If this does not work, check your UEFI firmware / BIOS settings for CPU frequency scaling settings.