deinstapel/cpupower #184: Periodic screen stutter when extension is enabled

feer9, 2022-02-06:
CPU: Intel Core i5-8400
GPU: AMD Radeon RX 470
OS: Manjaro Linux
Linux kernel: 5.15.16-1-MANJARO
Gnome-shell: 41.3-1
cpupower: version 26 installed from extensions.gnome.org

First of all, I'm not completely sure that this bug is only caused by this extension, but the issue manifests when I enable it.

What happens is, with a period of exactly 1 second, the entire screen stutters for about 50 ms (wild guess there).
I've noticed that this issue only manifests when running gnome in Wayland. Everything runs smooth in X11.

Here is a video running glxgears, check the awful frametime graph, and you can visually see the gears stop turning
https://youtu.be/KgS0cc01ZLc

Any additional info you need just tell me.

realKarthikNair, 2022-02-09:
Same issues here. 

> Using latest version of the extension from https://extensions.gnome.org/

  My system configs are as follows:
  
    Machine:   Type: Laptop System: ASUSTeK product: VivoBook_ASUSLaptop X421IAY_M413IA v: 1.0 
    
    System:    Host: cosmic Kernel: 5.15.15-76051515-generic x86_64 bits: 64 Console: tty pts/1 
               Distro: Pop!_OS 21.10 
    
    CPU:       Info: 6-Core model: AMD Ryzen 5 4500U with Radeon Graphics bits: 64 type: MCP cache: 
               L2: 3 MiB 
               Speed: 1397 MHz min/max: 1400/2375 MHz Core speeds (MHz): 1: 1397 2: 1397 3: 1806 
               4: 2154 5: 1397 6: 1958 
    
    Graphics:  Device-1: AMD Renoir driver: amdgpu v: kernel 
               Device-2: IMC Networks USB2.0 HD UVC WebCam type: USB driver: uvcvideo 
               Display: server: X.Org 1.21.1.2 driver: loaded: amdgpu,ati 
               unloaded: fbdev,modesetting,vesa resolution: 1920x1080~60Hz 
               OpenGL: renderer: AMD RENOIR (DRM 3.42.0 5.15.15-76051515-generic LLVM 12.0.1) 
               v: 4.6 Mesa 21.2.2 
    
    Drives:    Local Storage: total: 476.94 GiB used: 97.07 GiB (20.4%) 
               ID-1: /dev/nvme0n1 vendor: Intel model: SSDPEKNW512G8 size: 476.94 GiB

fin-ger, 2022-02-09:
Hi all! I'm sorry you are experiencing problems with this extension.

The issue is most likely related to the current frequency display. Try disabling the "Show current frequency" option in the settings. Let me know if this helps!

fin-ger, 2022-02-09:
Alright, I know how to fix this, but unfortunately I currently don't have the time to do this now. I'll mark this as urgent so it will be fixed with the next release.

realKarthikNair, 2022-02-10:
Thanks for the support. For the time being I will use slimbook battery module

fin-ger, 2022-04-30:
This needs some heavy lifting in the backend, I currently don't have the time to implement this. Unfortunately, this requires very deep knowledge of the internal parts of this extension, so I don't think anyone else can do it without investing a lot of time into this.

But thanks for the bump, maybe I'll code this on a train trip or sth.

fin-ger, 2022-04-30:
#162 is related to this.

NicolasGoeddel, 2022-10-02:
I've got the same issue here. I noticed it when I scroll evenly over a web page or something like that. Then it stuttered every second.
I am really interested in what can cause stutter the whole desktop from such an extension.

fin-ger, 2022-10-02:
> I am really interested in what can cause stutter the whole desktop from such an extension.

It seems that on certain CPUs the fetching of the current frequency of each core makes the CPU hang for a moment.

peterekepeter, 2025-03-29:
I discovered the same issue. I'm now disabling extensions when having stutters is unacceptable and then I re-enable it when I'm just browsing.

Disabling show current frequency helps. But then I have no idea what profile is currently on. It would be nice if the icon could reflect that.