deinstapel/cpupower #186: unable to open preferences on Gnome 40.5

crysman, 2022-03-31:
I've posted it already here
https://github.com/deinstapel/cpupower/issues/158#issuecomment-1080003994
Thanks for help!

fin-ger, 2022-04-01:
Executable bit is not set for `/usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js`. To (temporarily) fix this run:

```
$ sudo chmod +x /usr/share/gnome-shell/extensions/cpupower@mko-sl.de/src/prefs40/main.js
```

crysman, 2022-04-01:
resolved here https://github.com/deinstapel/cpupower/issues/158#issuecomment-1086133002

fin-ger, 2022-04-01:
No no no :D

Not resolved. This is quite a serious issue. Needs to stay open until someone (my time is currently very limited) fixes the issue.

fin-ger, 2022-04-01:
"Fixing" means updating the apt package.

crysman, 2022-04-01:
> No no no :D
> 
> Not resolved. This is quite a serious issue. Needs to stay open until someone (my time is currently very limited) fixes the issue.

Agree, sorry to be impetuous, glad you re-opened and glad I've found a potential serious bug ;)

Pear-sudo, 2022-08-25:
> 

I have the same problem on Gnome 42.2, Ubuntu 22.04.1 LTS ([following the official installation guide](https://github.com/deinstapel/cpupower#any-newer-ubuntu-release-or-ubuntu-based-distribution-if-unsure-do-this)). And this command works. Thank you!