deinstapel/cpupower #188: Add support for GNOME 42

MrSquaare, 2022-04-22:
The extension is labelled as not supported for GNOME 42 as this version is not included in supported versions.
I don't know if it will be compatible without changes

durzo, 2022-04-23:
+1, just installed Ubuntu 22.04 and there are no cpu freq monitors available. I used this for years on previous Ubuntu's and it would be great to have it again

durzo, 2022-04-23:
so I just discovered the ubuntu PPA for this package and installed it on 22.04 only to discover its missing "42" in "shell-version" of metadata.json - fixing that and chmod +x the prefs40/main.js and everything works.

ankurkotwal, 2022-04-24:
Submitted PR #189 for this issue.

fin-ger, 2022-05-13:
Quick summary why it's not merged, yet:

I want to test the RPM and DEB packages before a release. Further, the Fedora infrastructure suspended my copr Account as the name includes invalid characters (they messed up the registration process :rage:). I have to transition the Fedora/RedHat packages to a new Account and basically have to redo the whole infra setup as I cannot access the old config (login to my Account not allowed any more). Maybe I'll get in contact with the admins of fedoraproject.org. Next, I would like to include the new translations from Crowdin.

In summary, I just need to find some time to do all this.

Keltere, 2022-05-18:
Please is it possible to also add support for gnome 41 and if it's already supported could @martin31821 add it to the gnome extension page?

fin-ger, 2022-05-18:
@Keltere Gnome 41 is included. There is no release, yet. Also, the gnome extensions website will most likely be the last one to get the update. The release process is often very slow there.

Keltere, 2022-05-18:
> @Keltere Gnome 41 is included. There is no release, yet. Also, the gnome extensions website will most likely be the last one to get the update. The release process is often very slow there.

You mean there is a slow review process there or is martin not keeping up with the updates?

RafalSkolasinski, 2022-05-18:
Are they making problems like apple or play stores?

fin-ger, 2022-05-18:
> Are they making problems like apple or play stores?

I don't know. I do not deploy on those platforms.

I can understand why Gnome tries to enforce certain rules, the problem is the "when". The worst moment in a projects lifecycle is to enforce rules **after** a release has been made (and decisions and months of work are done).

Apart from that, I don't think I'll have enough time to deal with the Fedora Copr stuff until the weekend, and my weekend is already planned, so unfortunately no release this week :disappointed:

RafalSkolasinski, 2022-05-18:
That's completely fair, finding time for side projects is always difficult. 

Let's home the Gnome team won't make troubles once you get to it!

Zesko, 2022-05-22:
I edited `~/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/metadata.json` to add the version 42:
```json
 "shell-version": [
    "3.28",
    "3.36",
    "3.38",
    "40",
    "42"
  ],
```

Then reboot, it works fine in Gnome 42 like before old Gnome 40, 41 on the same device. But I do not notice any problem.

fin-ger, 2022-05-22:
Github actions currently has [broken apt-sources from azure.archive.ubuntu.com and lintian is not installable](https://github.com/deinstapel/cpupower/runs/6545963991?check_suite_focus=true). I updated the PPA, but the Gnome Extensions release has to wait until the CI passes.

I contacted a fedoraproject admin regarding my access to the cpupower copr.

So, debian, ubuntu and popos folks can update using the PPA. All other people have to wait or install from git (no github release available, yet).

tnfru, 2022-05-26:
Had to wipe my PC and looking to reinstall - after the merge I can't seem to install it.

When installed via apt it just doesn't show up in services, same with cloning the repo.

fin-ger, 2022-05-26:
I'm sorry you have trouble getting this to run. However, the release is not finished, yet. I think I can complete it this weekend.

Regarding your problem: you should generally stick with one installation method when installed once, unless you know what you are doing. Without a bit more detail on what you did (which actions in which order, also the commands you used to implement the actions) we can't help you that much.

What do you mean by not showing up in services? What services?

Cloning the repo will not install anything. This is solely for developers to have access to the extensions source and being able to apply changes to the source. It is not enough to "install" anything. If you have never heard of these things, please do not continue with this installation approach!

tnfru, 2022-05-26:
Sorry, I see I dind't provide enough information.
When I added the PPA and installed it via apt the service wasn't enabled and could not
be enabled because it was listed when running `gnome-extensions list `.

I tried:

> $ git clone https://github.com/ankurkotwal/cpupower.git
> $ cd cpupower
> $ make install PREFIX=/home/lars/.local

which resulted in the following error:

```
Compiling schemas...
Updating translations...
./scripts/update-translations.sh: line 9: /usr/bin/msgfmt: No such file or directory
make: *** [Makefile:49: build] Error 127
```

fin-ger, 2022-05-26:
Please follow the **exact** instructions in the README. I figure you forgot to logout and back in again. Please do not try to install the extension via the makefile if you have previously installed it via apt.

https://github.com/deinstapel/cpupower#any-newer-ubuntu-release-or-ubuntu-based-distribution-if-unsure-do-this

tnfru, 2022-05-26:
Yup I forgot to relog - after that I can activate it. But the indicator in the top bar does not show the current percentage and does not let me open preferences. I can't access preferences through Extension List either.

fin-ger, 2022-05-27:
Can you provide logs?

https://github.com/deinstapel/cpupower#viewing-log-output

Make sure to strip any sensitive information from the output.

Also, what is the output of

```
apt list gnome-shell-extension-cpupower
```

and

```
lsb_release -a
```

Also please check the version of the extension in the GNOME Extensions App.

tnfru, 2022-05-27:
`apt list gnome-shell-extension-cpupower` 
```
Listing... Done
gnome-shell-extension-cpupower/jammy,jammy,now 10.1.0-1~jammy1 all [installed]
```
and 

`lsb_release -a`
```
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy
```

journalctl output:

```
May 27 10:45:00 pop-os gnome-shell[2903]: Ignoring excess values in shadow definition
May 27 10:45:00 pop-os gnome-shell[2903]: Ignoring excess values in shadow definition
May 27 10:45:00 pop-os gnome-shell[2903]: Ignoring excess values in shadow definition
May 27 10:45:00 pop-os gnome-shell[2903]: Ignoring excess values in shadow definition
May 27 10:45:00 pop-os gnome-shell[2903]: Ignoring excess values in shadow definition
May 27 10:45:00 pop-os gnome-shell[2903]: Ignoring excess values in shadow definition
May 27 10:45:01 pop-os gnome-shell[2903]: Could not create transient scope for PID 20312: GDBus.Error:org.freedesktop.DBus.Error.UnixProcessIdUnknown: Process with ID 20312 does not exist.
May 27 10:45:03 pop-os gnome-shell[2903]: Failed to import DBusMenu, quicklists are not avaialble: Error: Requiring Dbusmenu, version none: Typelib file for namespace 'Dbusmenu' (any version) not found
May 27 10:47:20 pop-os gnome-shell[2903]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x260000b
May 27 10:49:29 pop-os gnome-shell[2903]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x260000b
May 27 10:49:48 pop-os gnome-shell[2903]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x260000b

```

fin-ger, 2022-05-29:
Unfortunately, I'm unable to reproduce your issue. I tried several times to setup a PopOS in a VM but the installer got stuck at ~14% during extracting files. The `sha256sum` is ok for the iso, so I don't know why it's not working.

I tried the installation from the PPA under debian 11 and Ubuntu 22.04 and both worked flawlessly.

You might have still some old installation in your users home folder or sth. You can try to follow these instructions to clear these installations: https://github.com/deinstapel/cpupower#uninstalling-1

Replace `username` with the username of your account on your PC.

fin-ger, 2022-05-29:
The ZIP for the Gnome-Extensions site is now ready and @martin31821 is currently uploading it for review. Further, the PPA is up-to-date (support for Bionic, Focal, and Jammy, as well as working with Debian Buster and Bullseye) and a new Copr has been created under https://copr.fedorainfracloud.org/coprs/fin-ger/cpupower with support for CentOS, EPEL, Fedora, Mageia, OpenMandriva, and OpenSUSE.

tnfru, 2022-06-17:
Hi, my issue was fixed by uninstalling the apt-package and reinstalling it through the gnome extension store. Works like a charm again :+1:

NicolasDerumigny, 2022-11-06:
This PR should be now closed (fixed by #189 )