deinstapel/cpupower #192: Not installable on ostree OSes

hyphone, 2022-06-23:
Hello and good day,

would you be interested to make this nice extension compatible with immutable OSes like Fedora Silverblue?

/usr for example is not writable and the udev rules for example could be written to /etc instead.

The script itself could still rely within the extension folder and does not need to be in /usr/bin or /usr/local/bin.

What do you think?

All the best
marco

fin-ger, 2022-06-24:
Afaik GNOME Shell extensions are unsupported on ostree by design. Therefore no GNOME shell extension is able to run on Silverblue. Please prove me wrong if that is not the case. Thank you for your contribution!

hyphone, 2022-06-24:
> Afaik GNOME Shell extensions are unsupported on ostree by design. Therefore no GNOME shell extension is able to run on Silverblue. Please prove me wrong if that is not the case. Thank you for your contribution!

This not quite true. I have installed several extensions installed.
Extensions are install-able by the user.

It's just that changes to the root filesystems can't be made without creating a layer.

But everything in userspace does not change at all.

hyphone, 2022-06-24:
Because /etc/ is still writable the udev rules could be moved to there and the executable could rely within the path of the extension itself instead to get moved to /usr or /local/usr, that is not writable in Silverblue.

hyphone, 2022-06-24:
In Silverblue and Kinoite the changeable things by the user are moved to /var (/var/etc, /var/home) and the previous paths (/etc and /home) are just symlinks to there. Everything in /var is changable.

fin-ger, 2022-06-24:
Having the cpufreqctl executable in a user-writable location is a no-go for security reasons as this script gets executed with superuser rights and must never be writable by an unprivileged user.

But nevertheless ostree support is then worth having a look at. Thanks for your hint!

hyphone, 2022-06-24:
What do you think about /opt/ :)

fin-ger, 2022-06-24:
If that's a location usable in ostree and not writable by normal users, it's fine, I think. I think the easiest approach would be to package this extension as a native package for ostree based distros. In other distros /opt is usually managed by the distros package manager and therefore not a good location to put the cpufreqctl binary. /usr/local is I think the only sane location for GNOME extension website installs.

hyphone, 2022-06-29:
I learned that it's indeed possible to use /usr/local/bin for custom executables and scripts.