deinstapel/cpupower #194: [Question] How to control from terminal?

RafalSkolasinski, 2022-07-15:
Just a question, the extension itself works as expected (and it's awesome BTW!). 

As the extension is doing really good job in controlling CPU freqs I wanted to use the same method to control them from terminal. 
I looked briefly through code and found references to `cpufreqctl` binary. 

However, when I try following
```bash
$ sudo cpufreqctl-rskolasinski info frequencies                                                                                                                                                             
mode: continuous
min: 17
max: 100
```
the output is slightly different from what I have currently set 
[image]

Question: what would be equivalent CLI command to read / set frequency limits?

RafalSkolasinski, 2022-07-15:
Also... where does the discrepancy come from?

fin-ger, 2022-07-16:
The info subcommand only outputs your hardware capabilities. Your CPU supports a minimum setting of 17% and a maximum of 100%.

Use `turbo get`, `max get`, and `min get` to get the current setting. Use `turbo set`, `max set`, and `min set` to set the values accordingly. Please keep in mind that the configured profiles are only available in the extension, not the cpufreqctl CLI tool.

fin-ger, 2022-07-16:
if you plan on scripting with the CLI tool, you might want to have a look at `--format=json` for easy output processing. But please keep in mind that this json API is not stable and might have different structure in a future update.

RafalSkolasinski, 2022-07-16:
Thank you so much, the

```bash
sudo cpufreqctl-rskolasinski max get
sudo cpufreqctl-rskolasinski max set 60
```
are doing exactly what I need :) I should have read the included `--help` probably more carefully! 

> if you plan on scripting with the CLI tool, you might want to have a look at --format=json for easy output processing. But please keep in mind that this json API is not stable and might have different structure in a future update.

Thanks, I'll keep it in mind. But mostly I just need it to "limit" CPU on some of headless boxes if they get too loud. 
And as your gnome extension is working reliably and doing exactly what I need it to normally I did not want to look for another solution.

closed 2022-07-16 by RafalSkolasinski