deinstapel/cpupower #199: Add support for Gnome 43

aneverhopefulhelper, 2022-09-27:
Hiya, could you update this so that it is available/works on Gnome 43 please? As this is marked as incompatible with it currently. Thank you!

sezcl, 2022-09-28:
+1 . Thank you!

vanastasis, 2022-09-29:
+1 please

Mimoja, 2022-10-17:
Non complete list of things not working with Gnome 43 / Ubuntu 22.10 by simply changing the metadata:

- Tray Icon (its just blank)
- Current Frequency (always .8 GHz, even tho `/sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq` suggests otherwise with AMD ryzen)

Ubuntu 22.10 also brings amd p_state which seems to work fine (also confirmed with 5.17 OEM kernel on 22.04)

nietzschescodes, 2022-10-25:
yeah, that would be nice and really useful to have a version for Gnome 43. is there only one person on that project?

fin-ger, 2022-10-25:
> yeah, that would be nice and really useful to have a version for Gnome 43. is there only one person on that project?

yes, currently I'm the only one working on this. I cannot give any ETA for an update. I currently have more important stuff to do. A realistic time to start working on this would be in December...

nietzschescodes, 2022-10-25:
thanks for your response. that is sad because your program is the only one that actually works on Gnome and it works perfectly. I don't get why Gnome includes their own power manager and it literally doesn't work on any of their gnome 4X.
anyway, thanks for your great program. In the meantime, I might add another DE. They might have a good power manager. I know Cinnamon has one, but it never worked or freeze the laptop. Might try maybe KDE. Can't wait for your Gnome 43 version!

fin-ger, 2022-10-25:
If you are running an Intel CPU you might have a look at [cpupower-gui](https://github.com/vagnum08/cpupower-gui). It is a DE-independent graphical program for managing Intel p-state. It does not have Shell integration like this extension, but might serve your needs. And most importantly it is independent of Gnome-Shell update shenanigans.

nietzschescodes, 2022-10-25:
Ok, thanks. I will try that before adding another DE.

NicolasDerumigny, 2022-11-03:
I have looked a bit and this is due to Gnome 43 changing the location of the battery toggle used for detection of the charging / not charging state. It was moved from `aggregateMenu` to `Main.panel.statusArea.quickSettings`. I found the source code [here](https://gitlab.gnome.org/GNOME/gnome-shell/-/blob/main/js/ui/panel.js) but is seems to differ quite a bit, as `_power` do not exist anymore, but `powerToggle` does, which seems not to have `_proxy.State`.

NicolasDerumigny, 2022-11-03:
Support is back again on my PR :).

nietzschescodes, 2022-11-03:
that would be great to have that extension upgraded, Nicolas! I have found a temporary solution, but it is not as good as this CPU Power Manager.

HydeFromT70s, 2022-11-03:
> Support is back again on my PR :).

Thanks @NicolasDerumigny for finding solution. I can confirm that making changes like in your PR and restarting Gnome Shell brings extension back to live in Gnome 43 :) Thanks!

nietzschescodes, 2022-11-04:
I don't get it. what we have to do to have it bring back to life? downloading the system.js created by nicolas  and put it where exactly? I downloaded it and put it in the file cpupower (in extensions in .locals) and nothing happened. what then should we do with that file? where we have to put it? do we have to remove a file in adding that file? sorry, I don't get what to do with that system.js file..

HydeFromT70s, 2022-11-04:
> I don't get it. what we have to do to have it bring back to life? downloading the system.js created by nicolas and put it where exactly? I downloaded it and put it in the file cpupower (in [...]

It's fairly simple. All you have to do is modify two files as you can see in @NicolasDerumigny PR.

Just edit file: /home/<YOUR_USERNAME>/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/src/indicator.js and change line 111 from:
`this.power = Main.panel.statusArea["aggregateMenu"]._power;`
to
`this.power = imports.ui.main.panel.statusArea.quickSettings._system._systemItem._powerToggle;`

Also you need to modify file: /home/<YOUR_USERNAME>/.local/share/gnome-shell/extensions/cpupower@mko-sl.de/metadata.json by adding your Gnome version to "shell-version":
 before: `"shell-version": ["3.28", "3.30", "3.36", "3.38", "40", "41", "42"],`
after: `"shell-version": ["3.28", "3.30", "3.36", "3.38", "40", "41", "42", "43"],`

After modifying this two files, you need to log out or restart your Gnome Shell by ALT+F2 typing r and hitting enter.

nietzschescodes, 2022-11-04:
Only those 2 small steps + log-out? It is quite simple. Thanks for explaining, but unfortunately for me, it doesn't work. When I check on my list of extensions, it still not showing that extension so I can't turn it on.

NicolasDerumigny, 2022-11-04:
It depends how you installed the extension first. If it was done by the package manager, then the files are located in `/usr/share/gnome-shell/extensions/[cpupower@mko-sl.de](mailto:cpupower@mko-sl.de)/`. Or wait until this PR got merged :).

nietzschescodes, 2022-11-04:
Hi Nicolas. Yes, I had find the file and made the changes that Hyde was talking about, but I don't know how to turn it on, because it does not appear back on my list of the extensions (see pic below). But no worry, I will wait until this PR got merged. Thanks so much for your work!
[image]

NicolasDerumigny, 2022-11-05:
Hi folk!
I have started looking a bit more into the code and updated it (not to break compatibility with older GNOME versions....).
Now we need you help. I am running under Arch Linux and cannot test on other distributions, but (copy/paste from the GNOME 42 update):

For such a version increment to happen, we need testing in the most popular Linux distributions, including

all supported Ubuntu LTS releases
- [ ] 22.04 (Gnome 42)
- [ ] 20.04 (Gnome 3.36)
- [ ] 18.04 (Gnome 3.28)

latest debian testing, stable, and old-stable
- [ ]  testing (bookworm) (Gnome 42)
- [ ] stable (bullseye) (Gnome 3.38)
- [ ] old-stable (buster) (Gnome 3.30)

Fedora stable and old-stable

- [ ] Fedora 34 (soon EOL, maybe skip?) (Gnome 40)
- [ ] Fedora 35 (Gnome 41)
- [ ] Fedora 36 (Gnome 42)
- [x] latest Arch (Gnome 43)
- [ ] latest PopOS (22.04, Gnome 42)

# Testing procedure

Run

$ make install PREFIX=$HOME/.local
$ dbus-run-session -- gnome-shell --nested --wayland

and check for errors and warnings regarding cpupower in the console output while doing the following steps:

1. Do clean installation (via the dropdown)
2. Moving both sliders in the dropdown with
        i. Mouse
        ii. Arrow keys
        iii. Mouse wheel
3. Toggle Turbo Boost
4. Change Profile
5. Toggle Auto-Switch (testing maybe not possible as VMs don't fire AC adapter events)
6. Open Preferences and toggle all options
7. Do 2-4 for all available backends
8. Create a new profile and change its position (move up and down in the list)
9. Set all values in that new profile
10. Select it in the extensions dropdown menu
11. Remove the profile
12. Open About Window
13. Uninstall tool and polkit rules (this is known to be flaky, if this fails this is not a stopper)
14. Install latest version from Gnome Extensions
15. Update with this version
16. Check if update procedure is working (Note: this requires a version increment which is only done by `make release` which I have not done yet, so version in `installer.sh` and `cpufreqctl` needs to be incremented manually for testing)
17. Native installs
    i. For debian-based distros
            a. do a clean install of the latest .deb for this extension
            b. build a deb for the new version and update with the package manager and check if the update is working
     ii. For red-hat based distros
            a. do a clean install of the latest .rpm for this extension
            b. build a rpm for the new version and update with the package manager and check if the update is working

Global installation method and all other installation methods are not tested (at least by me). You can do it if you want and if you encounter any issues, I will do my best to fix them.

Orca (screen reader) support can only be tested in "real" gnome-shells (non-nested). At least I couldn't get Orca running in the nested gnome-shell, but it would be nice to have basic testing for screen-readers in all distros (profile management in preferences dialog is known to be hard-to-impossible to use with orca, this will be fixed when redesigning the preferences dialog and not a stopper for a new release).

nietzschescodes, 2022-11-05:
Just a small heads up: Testing (bookworm) comes with Gnome 43 since September.

nietzschescodes, 2022-11-06:
Looks you can skip PopOS. They skipped 22.10 because it seems they don't want Gnome 43 and are making their own DE.

NicolasDerumigny, 2022-11-06:
I don't think PopOS! should be skipped, as long as GNOME is still supported by default on this distro (which is the case, right?)

fin-ger, 2022-11-06:
> I don't think PopOS! should be skipped, as long as GNOME is still supported by default on this distro (which is the case, right?)

Yes, if PopOS skipped a release, the latest release is still the one that needs testing.

nietzschescodes, 2022-11-06:
I believe Gnome 42 is the default on PopOS 22.04 but they have decided to stop using Gnome for the future versions, that is why they won't release a 22.10. They don't want Gnome 43 and instead are working deeper on their own DE for the 23.04 edition. But right, ofc, the new extension should still be compatible with their Gnome 42.

nietzschescodes, 2022-11-06:
Oh, I just noticed something else. Gnome 3.36 is shipped on Ubuntu 20.04, not 3.30. Also I think it is Gnome 3.28 that is shipped on Ubuntu 18.04, not 3.22.

fin-ger, 2022-11-06:
> Oh, I just noticed something else. Gnome 3.36 is shipped on Ubuntu 20.04, not 3.30. Also I think it is Gnome 3.28 that is shipped on Ubuntu 18.04, not 3.22.

Feel free to correct this if it's wrong. I looked this up with the packages provided for `gnome-session` in the official repositories of each distro without backports.

NicolasDerumigny, 2022-11-06:
Good call. I was looking at the `gnome` package which does not exactly match.

nietzschescodes, 2022-11-08:
Since I have tried the tricks given here many times and it doesn't work for me on Gnome 43 or appears as not compatible even if the changes were done, can someone who has a functional CPU Power Manager gives me, please, the setting "Quiet" parameters? What is the minimum in MHz and what is the maximum in MHz? I would just adjust the settings manually like with cpufrequtils or something like that.

LeSnake04, 2022-11-21:
> If you are running an Intel CPU you might have a look at [cpupower-gui](https://github.com/vagnum08/cpupower-gui). It is a DE-independent graphical program for managing Intel p-state. It does not [...]

PLEASE DO NOT INSTALL MASTER ON FEDORA 36 !!! https://github.com/vagnum08/cpupower-gui/issues/96

nietzschescodes, 2022-11-21:
I did try cpupower-gui when I was suggested that in this thread. It just doesn't work. You set a limit and the limit is crossed.
My solution after was simply to install cpufrequitls and set manually what I want and it works perfectly.

lp35, 2022-12-01:
Hi @NicolasDerumigny !

Thanks for giving us a complete test case for validating this.

I have run and passed successfully all those tests on Fedora 37! You can tick this distro in your list ;)

Thanks!

NicolasDerumigny, 2022-12-01:
Thanks a lot for your testing, I update! The testing procedure is from @fin-ger , I have just pasted it from another thread :).

nietzschescodes, 2023-01-31:
So did the extension has been updated properly on the site of gnome extensions now? I had to revert to Gnome 3.38 so I can't see it if it is compatible with Gnome 43 . Oh and just a heads up, Gnome 44 will be released soon. In March.

NicolasDerumigny, 2023-01-31:
I have not touched this project, so I do not think this is the case, as further testing was needed for a merge.

orvitpng, 2023-03-04:
@NicolasDerumigny have you gotten around to this? Gnome 43 is now out on arch.

NicolasDerumigny, 2023-03-11:
Sorry, I have been busy with another project, but now that my machine has been updated, expect a PR soon :).

LeSnake04, 2023-03-12:
Just FYI, gnome 44 will come out on the 22.03. 
So it will be likely available to most Distros around middle to end of April (and to arch at the start of April)

NicolasDerumigny, 2023-05-07:
There is my support for GNOME 44 https://github.com/NicolasDerumigny/cpupower :-).

NicolasDerumigny, 2023-05-07:
Note that I haven't had time to rewrite the `cpufrectl` utility in JS (the original messes up the measurements on my system as is triggers boost) and likely won't until at least summer, so there is no way this gets eventually published here as it depends on my (custom) modification the former.

nietzschescodes, 2023-06-10:
Debian 12 Stable is released today. It has Gnome 43 as a DE. Is the gnome extension now compatible with Gnome 43?