deinstapel/cpupower #208: Support for Gnome 44

mokilcde, 2023-03-09:
# Creating a new issue

* Check if another issue with the same problem exists
No
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
     Yes
  * policykit working and rule working
    Yes
* Check if the cpufreqctl utility works correctly
  Yes
* Please include the following information (only for bugreports):
  * CPU Model
  * Linux kernel version
  * policykit version
  * Gnome shell version
    See below
  * journalctl -xb (maybe pastebin/filtered)

**To Reproduce**
I did only an update from 43.x to 44.rc

**System**
```
-- System Information:
Debian Release: 12.0
  APT prefers unstable
  APT policy: (710, 'unstable'), (670, 'testing'), (660, 'experimental'), (600, 'stable'), (500, 'stable-security'), (500, 'oldstable')
Architecture: amd64 (x86_64)
Foreign Architectures: i386, armhf, arm64

Kernel: Linux 6.1.0-6-amd64 (SMP w/8 CPU threads; PREEMPT)
Locale: LANG=en_GB.UTF-8, LC_CTYPE=en_GB.UTF-8 (charmap=UTF-8), LANGUAGE=en_GB:en
Shell: /bin/sh linked to /usr/bin/dash
Init: systemd (via /run/systemd/system)
LSM: AppArmor: enabled

Versions of policykit:
ii  policykit-1                                      122-3
ii  policykit-1-doc                                  122-3
ii  policykit-1-gnome                                0.105-8

Versions of packages gnome-shell depends on:
ii  dconf-gsettings-backend [gsettings-backend]  0.40.0-4
ii  gir1.2-accountsservice-1.0                   22.08.8-6
ii  gir1.2-adw-1                                 1.3~rc-1
ii  gir1.2-atk-1.0                               2.47.1-1
ii  gir1.2-atspi-2.0                             2.47.1-1
ii  gir1.2-freedesktop                           1.75.6-1
ii  gir1.2-gcr-3                                 3.41.1-1+b1
ii  gir1.2-gdesktopenums-3.0                     44~beta-1
ii  gir1.2-gdkpixbuf-2.0                         2.42.10+dfsg-1+b1
ii  gir1.2-gdm-1.0                               43.0-3
ii  gir1.2-geoclue-2.0                           2.6.0-2
ii  gir1.2-glib-2.0                              1.75.6-1
ii  gir1.2-gnomebg-4.0                           44~beta-1
ii  gir1.2-gnomebluetooth-3.0                    42.5-3
ii  gir1.2-gnomedesktop-4.0                      44~beta-1
ii  gir1.2-graphene-1.0                          1.10.8-1
ii  gir1.2-gstreamer-1.0                         1.22.0-2
ii  gir1.2-gtk-4.0                               4.10.0+ds-2
ii  gir1.2-gweather-4.0                          4.2.0-2
ii  gir1.2-ibus-1.0                              1.5.28-1
ii  gir1.2-mutter-12                             44~rc-1
ii  gir1.2-nm-1.0                                1.42.2-2
ii  gir1.2-nma4-1.0                              1.10.6-1
ii  gir1.2-pango-1.0                             1.50.12+ds-1
ii  gir1.2-polkit-1.0                            122-3
ii  gir1.2-rsvg-2.0                              2.54.5+dfsg-1
ii  gir1.2-soup-3.0                              3.3.1-1
[... 171 lines cut ...]
ii  libx11-6                   2:1.8.4-2
ii  libx11-xcb1                2:1.8.4-2
ii  libxau6                    1:1.0.9-1
ii  libxcb-randr0              1.15-1
ii  libxcb-res0                1.15-1
ii  libxcb1                    1.15-1
ii  libxcomposite1             1:0.4.5-1
ii  libxcursor1                1:1.2.1-1
ii  libxdamage1                1:1.1.6-1
ii  libxext6                   2:1.3.4-1+b1
ii  libxfixes3                 1:6.0.0-2
ii  libxi6                     2:1.8-1+b1
ii  libxinerama1               2:1.1.4-3
ii  libxkbcommon-x11-0         1.5.0-1
ii  libxkbcommon0              1.5.0-1
ii  libxkbfile1                1:1.1.0-1
ii  libxrandr2                 2:1.5.2-2+b1
ii  libxtst6                   2:1.2.3-1.1
ii  mutter-common              44~rc-1

Versions of packages gnome-shell is related to:
ii  libegl-mesa0 [libegl-vendor]  23.0.0-1
ii  libgl1-mesa-dri               23.0.0-1
ii  libglx-mesa0 [libglx-vendor]  23.0.0-1
```

Lawouach, 2023-04-24:
That would be so great if it could be updated indeed. It's probably the only extension I've kept through the years :)

gcqmkm02, 2023-05-05:
Oui, s'il vous-plaît!