deinstapel/cpupower #214: Support for G45 version

gilvbp, 2023-08-11:
Hi, could you update to G45 version?
This tutorial might help:
https://gitlab.gnome.org/ewlsh/gjs-guide/-/blob/d4cb8a9fbdf669c791dc24bc1b052e404779157b/docs/extensions/upgrading/gnome-shell-45.md
Thanks.

# Creating a new **issue**

* Check if another issue with the same problem exists
* Check if you got the prerequisites right
  * pstate driver enabled and supported CPU
  * policykit working and rule working
* Check if the cpufreqctl utility works correctly
* Please include the following information (only for bugreports):
  * CPU Model
  * Linux kernel version
  * policykit version
  * Gnome shell version
  * journalctl -xb (maybe pastebin/filtered)

schlagmichdoch, 2023-09-15:
I'd appreciate this too!

gilvbp, 2023-09-26:
Opened: https://github.com/deinstapel/cpupower/pull/215, but still some files to upgrade to [ESM](https://github.com/gilvbp/cpupower/commit/a55c098897729e25fe3e27611c1b2ceccc564d9d).

Neo-29, 2023-10-15:
How this evolving?
a way to make it work on G45?

gilvbp, 2023-10-16:
Didnt evolve. I didn't have time to work on it.

Neo-29, 2023-10-16:
> Didnt evolve. I didn't have time to work on it.

Any plan on the future.?

eleius, 2023-10-25:
> Any plan on the future.?

Unfortunately he left linux and stopped worked on it ([source](https://github.com/deinstapel/cpupower/pull/215#issuecomment-1773770769))

You can use this for now: https://github.com/DisDis/cpufreq/tree/support_gnome45

NicolasDerumigny, 2023-11-08:
I tried to update the code to work on GNOME 45. Still very early support, with:
- sliders not working
- preference page not working

Basically you can only enable / disable turbo and switch between profiles if you had one saved before. I do not know if / when I have time to dig up some more, but that's better than nothing. Any help is more than welcome.

Code is available at https://github.com/NicolasDerumigny/cpupower

BloodyIron, 2023-11-20:
> > Any plan on the future.?
> 
> Unfortunately he left linux and stopped worked on it ([source](https://github.com/deinstapel/cpupower/pull/215#issuecomment-1773770769))
> 
> You can use this for now: https://github.com/DisDis/cpufreq/tree/support_gnome45

Thanks for this! :) (the link to the G45 fork)

gdlin, 2024-01-05:
> I tried to update the code to work on GNOME 45. Still very early support, with:
> 
> * sliders not working
> * preference page not working
> 
> Basically you can only enable / disable turbo and switch between profiles if you had some saved before. I do not know if / when I have time to dig up some more, but that's better than nothing. Any help is more than welcome.
> 
> Code is available at https://github.com/NicolasDerumigny/cpupower

hey @NicolasDerumigny, thanks for your work; I've managed to download and install it. I wanted to open an issue but they are disabled, just wanted to let you know that if you create a new tag on your repo, it **should** create a new release automatically, which can be installed easily using the gnome system. Github actions is free for public repos.
Maybe you can point the users to the installation steps in https://github.com/NicolasDerumigny/cpupower?tab=readme-ov-file#your-distribution-is-not-listed

Thanks again.

PS: @fin-ger maybe you can check this code out since it seems to be working good and there are no merging conflicts. thanks a lot!

NicolasDerumigny, 2024-01-08:
I stopped working on that because of other projects (and that I do not know anything about GUI programming in GJs) ; and I would not be comfortable publishing a tag with a bug as bad as "no GUI"...

I use an arch pkgbuild to install it on my system with modified url to point to my repo, but this is not designed to be a long-term solution.

I have re-activate issues if you want to open some / create PRs :)