deinstapel/cpupower #220: No Release file for Ubuntu 24.04 ?

p-phillips, 2024-06-30:
I just tried to install on a 24.04 install and got bitten by the Release file issue:

```
$ sudo add-apt-repository ppa:fin1ger/cpupower
[sudo] password for xxxx: 
Repository: 'Types: deb
URIs: https://ppa.launchpadcontent.net/fin1ger/cpupower/ubuntu/
Suites: noble
Components: main
'
Description:
Manage the frequency scaling driver of your CPU (Intel Core and AMD Ryzen processors supported).
More info: https://launchpad.net/~fin1ger/+archive/ubuntu/cpupower
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
...
Reading package lists... Done         
E: The repository 'https://ppa.launchpadcontent.net/fin1ger/cpupower/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

```

Is this because it's not been ported to 24.04 yet or due to an oversight in the package management ?

Pete

p-phillips, 2024-07-01:
Just realised this is a different version of cpupower.  Therefore closing it

fin-ger, 2024-07-01:
Yes, this extension has not been ported to new Gnome versions for a long time. Therefore there also does not exist any version for newer Ubuntu versions. I am experimenting with some tooling based on the Meson build system for automating end2end tests for doing automated compatibility testing (which is the most painful thing for an update). But don't get your hopes too high for a new version any time soon. Maintaining a Gnome shell extension is a liability. It is almost guaranteed to brick with every new update, so I'm not that eager to "fix" it for the next 2-3 months to start over again :(

p-phillips, 2024-07-01:
OK - thanks for replying anyway - much appreciated.  
Pete

closed 2024-07-01 by p-phillips